prusa3d/MM-control-01 #23: Questions: MMU2 contoller on MK2.5 miniRAMbo ???

3d-gussner, 2018-08-30:
Q1: Did you test the MMU2 controller on the MK2.5 miniRAMbo?
- P3 connector pin 9 on an Einsy is 'PJ5' (pin76) but on the miniRAMbo this pin is 'reset'.
  - I think there is a need for a y-cable for the MMU2 signal cable, similar to the MK2.5 P.I.N.D.A. v2 and Noctua.
  - Opened a pull request for Prusa-firmware https://github.com/prusa3d/Prusa-Firmware/pull/1024

Q2: As the MK2.5 runs on 12V vs. MK3 24V, do you need to adjust the power/current/settings for the MMU2 TMC2130s?

mkbel, 2019-02-23:
Q2: We use stealth chopper on 12V printer, as the settings for 24V works fine, but produce terrible noise.

mkbel, 2019-02-25:
Q2: Yes. https://manual.prusa3d.com/Guide/3.+Electronics+and+MMU2+unit+assembly/757?lang=en#s17813

closed 2019-02-25 by mkbel