prusa3d/MM-control-01 #24: Bootloader Missing

CacikeOtoao, 2018-09-18:
Can you please specify which bootloader to use on the Controller. It is missing from the documentation.

CacikeOtoao, 2018-09-19:
@akukan Can you please specify which Bootloader is used on the MMU board? And will you guys make it available as part of the GPL OSHW?

akukan, 2018-09-20:
@CacikeOtoao we will release it in next few days, sorry for delay

CacikeOtoao, 2018-09-20:
@akukan  I look forward to the prompt release, also a good set of instructions regarding how to set up the environment and compile steps would be welcomed. Glad you're taking serious steps with GPL compliance.

CacikeOtoao, 2018-09-23:
@akukan any updates?

tito2018, 2018-09-24:
@CacikeOtoao have they addressed this?

tito2018, 2018-09-24:
@akukan are you guys going to respond and make the  bootloader hex file available? There seems to be an interest for this to be available.

oemelectronix, 2018-09-24:
@CacikeOtoao @tito2018 I've also asked them on support chat and through email about the bootloader but they have not resolve the issue yet. Hopefully since more people are asking for it they will provide it.

CacikeOtoao, 2018-09-24:
Glad there are more guys looking for the bootloader, I asked a few days ago but is not available yet.

CacikeOtoao, 2018-09-24:
@akukan did you guys locked the fuse bits? Are we able to read the chip? The lack of information is getting a bit frustrating.

tito2018, 2018-09-24:
@akukan @oemelectronix  @CacikeOtoao I've tried for the past few days to load the firmware on my MMU2 board built from their kicad source files and the firmware is not doing anything. Have also asked for the Bootloder and they are just ignoring the request.

CacikeOtoao, 2018-09-25:
@akukan I really wished you guys followed Lulzbot lead on Open Source and released all Source code including bootloaders, instead of bits and pieces.

tito2018, 2018-09-26:
Guys I think the developer has been kidnapped. He disappeared 7 days ago.

mkbel, 2018-10-04:
Bootloader should and will be there: https://github.com/prusa3d/caterina
Sadly person with with rights to rewrite history is still on vacation. Now source there is not actual, missing build instructions and LUFA library sub repository doesn't point to prusa controled github repository.
In the meantime before it is rectified you can get actual bootloader source there:
https://github.com/mkbel/caterina/

Bootloader binary is shipped with the board definition for Arduino.
Additional boards manager URL:
https://raw.githubusercontent.com/prusa3d/Arduino_Boards/master/IDE_Board_Manager/package_prusa3d_index.json
Install Prusa Research AVR Boards by Prusa Research
which contains only one board:
Original Prusa i3 MK3 Multi Material 2.0

closed 2018-10-04 by mkbel

tito2018, 2018-10-04:
@mkbel @dronecz Thanks a lot guys. I was able to flash it using the IDE and Prusa Board definitions with my USBtiny.

mkbel, 2018-10-05:
@tito2018 I just wonder why you need to flash bootloader. Did you manage to brick MMUv2 board with bootloader from Prusa?

tito2018, 2018-10-05:
@mkbel I built a board based on the Git kicad documentation and had to burn the bootloader on it.

mkbel, 2018-10-09:
@tito2018 Thank you for your prompt reply.

mkbel, 2018-10-09:
Bootloader on https://github.com/prusa3d/caterina is up to date now.

jgalak, 2020-09-11:
Does anyone know if this procedure should work with the Atmel ICE programmer?  As I understand it, the only difference should be that the programmer is selected in the IDE under Tools->Programmer, rather than under "Port", but all I'm getting is "Error while burning bootloader."  The programmer has the right LEDs come on (middle is red, left is green) but that's the result I get....

mkbel, 2020-09-11:
@jgalak: Looks like you are trying to flash using JTAG interface, but this interface is not exposed on PCB. You need to use ICSP -  Serial Programming. I don't know if your programmer is capable of that.

mkbel, 2020-09-11:
@jgalak: If your programmer doesn't support ICSP you can buy USBASP AVR programmer from China for about 1.5 USD including shipping.

jgalak, 2020-09-11:
Thanks, that makes a lot of sense.  The Atmel ICE definitely supports ICSP as well as JTAG, but I'm not sure how to get the Arduino IDE to put it into ICSP mode.  I think I'll ask around the Arduino forum.  Or just set up a spare Arduino as a programmer.

jgalak, 2020-09-11:
Alternatively, if anyone knows how to flash the bootloader using Atmel Studio, I know that the programmer mode can be set when flashing with it.

jgalak, 2020-09-11:
I am still looking for info on how to get Arduino IDE to put the programmer into ICSP mode, but there's a second approach that might work:  burn it with Atmel Studio.  The main issue there is that I really don't want to try to compile the bootloader from source.  

The link given above (https://github.com/mkbel/caterina/) seems to point to a source repository, only.  Is the bootloader available anywhere as a pre-compiled .hex file?

Additionally, once the bootloader is flashed, I would need to set the fuses on the chip.  Are the fuse settings available somewhere?

Thanks.

mkbel, 2020-09-12:
@jgalak Precompiled bootloader is part of board definition. It is already mentioned here twice. https://raw.githubusercontent.com/prusa3d/Arduino_Boards/master/IDE_Board_Manager/package_prusa3d_index.json https://raw.githubusercontent.com/prusa3d/Arduino_Boards/master/IDE_Board_Manager/prusa3dboards-1.0.2.tar.bz2 inside that board definition archive in bootloaders/prusa_mm_control/Caterina-prusa_mm_control.hex

jgalak, 2020-09-12:
> @jgalak Precompiled bootloader is part of board definition. It is already mentioned here twice. [...]

Thank you!  I missed that somehow, but I did find it eventually, and was able to burn the bootloader successfully!  In case anyone else needs it, here's what I did:

1) I ended up using Atmel Studio 7.  It's free from the Microchip/Atmel website, and installing it also installs the drivers for the Atmel ICE programmer.
2) I added the board definition to Arduino IDE as explained above, and then on my Windows 10 machine went to: `C:\Users\USERNAME\AppData\Local\Arduino15\packages\PrusaResearch\hardware\avr\1.0.3\bootloaders\prusa_mm_control` where I found the file `Caterina-prusa_mm_control.hex`
3) I connected the programmer's 6 pin cable to the P5 header on the MMU2 board (pin one is a little hard to make out, but if you orient the cable so that the tab on the cable is facing away from the stepper connectors, it'll be correct) and the other end to the AVR port on the programmer.  Programmer is connected to the PC via a micro USB cable, but it does not power the board, you need to apply power in another fashion - I just connected it to the printer as per instructions and turned the printer on.  Hardware connection is good when the middle light on the programmer is red (programmer is powered) and the left light is green (board is powered and connected).  Right light can be solid green, blinking green, or off, depending on what program is already on the chip.
4) In Atmel Studio, go to Tools->Device Programming.
5) In the Tool drop down select Atmel-ICE, under Device select ATmega32U4, and under Interface select ISP.  Click Apply.  You can verify that all is good by clicking the "Read" button.  You should get a device signature and a target voltage back if the programmer is talking to the chip.
6) On the left side, select the Memories tab.  In the middle, under the label "Flash (32KB)", click the button with three dots, and select the bootloader file at the location mentioned in #2 above.  I left all other defaults as they were (Erase device before programming, and Verfy Flash after programming both checked), and clicked "Program".  After a few seconds, it came back with three messages in the bottom pane - Erasing device OK, Programming Flash OK, and Verifying Flash OK.
7) Last step is to set the fuses.  The same directory where the hex file was found contains a file called `program.txt`.  The first line of it has avrdude programming instructions for a different programmer, but includes the fuse settings: Extended - 0xCB, High - 0xD8, Low - 0xFF.
8) Select the Fuses tab on the right. enter those values into the Value boxes for each Fuse Register, leave everything else on default (Auto read and Verify after programming checked), and hit Program.  You should get an OK message for each register and then one for Verify registers.

That should be everything needed to accomplish this process.