prusa3d/MM-control-01 #27: Implement Power Saving

awigen, 2018-09-27:
The MMU produces quite a hissing sound while idle.
The stepper drivers should enter a reduced power mode while not moving.

AbeFM, 2018-10-02:
And the motors are always good, even just sitting there.

Elsewhere I saw a complaint of the selector drum losing it's orientation but getting hot and slipping over the stepper's shaft.

pokeimon, 2018-10-06:
I understand why the stepper drivers are powered, and I agree that they should at least enter a lower power state, but another possible solution would be that when the printer is idle the mmu steppers would also be disabled but when the printer homes/starts a print it also homes the mmu steppers.

KarlZeilhofer, 2018-10-10:
This guy here reported, that the temperature of the motor shaft was to hot, so the idler body melted and slipped on the shaft: https://www.youtube.com/watch?v=U3v_FSRmHss

I think Trinamic's "cool step" (TM) should be implemented/activated here!

AbeFM, 2018-10-10:
My Prusa is now the loudest printer in the house. The "fan noise" on just
the motors and drivers cooking.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

AbeFM, 2018-10-17:
From my experience, the hub loses itself from time to time anyway. So even if it had to rehome it wouldn't be a problem. It can be done pretty non-invasively.

mullcom, 2022-07-01:
Main problem.

Motor is 🔥 hot and active. 

Why? 
It says that it's need to have track on the position.

Solution.
Why not use a home calibration when it it's need to before the print or when the situation needed?

mullcom, 2022-07-01:
> Main problem.
> 
> Motor is 🔥 hot and active. 
> 
> 
> Why? 
> It says that it's need to have track on the position.
> 
> Solution.
> Why not use a home calibration when it it's need to before the print or when the situation needed?

So when i notis this issue i have not power off prisa printer everytime i have done a print so solve the heat and Wear on stepper motor