prusa3d/MM-control-01 #31: MMU2 fails to load filament on 1st print after exiting service menu

mylife4aiurr, 2018-10-02:
I think I posted in the wrong branch:
https://github.com/prusa3d/Prusa-Firmware/issues/1190

posting this here as this is correct place for mmu2 firmware issues
https://manual.prusa3d.com/Guide/Service+menu+-+bowden+length/821?lang=en
After doing bowden length calibration. U exit service menu by hitting reset on mmu.

So goto sd card, choose a print. Printer will begin heating up nozzle and bed, then do bed calibration. Once bed calibration is done, it fails to load filament.

Solution 1 (tried it)
I have to reset printer by hitting reset on mk3. Then print from sd card again.

Solution 2 (haven't tried it)
Just print from sd card a 2nd time. (Chris Riley's, live stream build of mmu2)

mkbel, 2018-10-04:
This is already fixed by https://github.com/prusa3d/MM-control-01/pull/28. As you can exit improved bowden length correction without MMU reset.

closed 2018-10-04 by mkbel

mylife4aiurr, 2018-10-06:
We shall see when new firmware in released