prusa3d/MM-control-01 #32: Filament grinds on load due to differing hobbed gear diameters

AbeFM, 2018-10-02:
There's lots of talk elsewhere about the topic, but it's unmentioned here: differences in hobbed gear size lead to misfeeding of material.

If the first (calibrated) channel is perfect, the second gear is small, you'll end up mm/cm short of primary extruder gears, and get a missed feed.

If you turn up the calibration distance from the "secret menu" to compensate, then you grind filament on the channels with a bigger gear, filling MMU with dust and shavings, and notching filament which causes missed feeds later in the printing process.

This accounts for 60% of my failures.

murphy2712, 2018-10-02:
They added a setup for each filament: https://github.com/prusa3d/MM-control-01/commit/4d7d513642b35015fc940da65c972913a9352197

jonbet83, 2018-10-02:
It would also be good if the bondtech gear turned to accept the filament load also, it makes no sense loading into static gears, even if each loading was a bit longer it would stop the grinding also

AbeFM, 2018-10-03:
I was thinking about that, but it has issues too, much as I like the idea.

If you don't know where the end of the filament is, you'll end up over-feeding the nozzle, get a blob, and all sorts of other baddies.

However taking 4 stabs at the optical sensor and getting something useful from there might work.

mkbel, 2018-10-04:
> It would also be good if the bondtech gear turned to accept the filament load also, it makes no sense loading into static gears, even if each loading was a bit longer it would stop the grinding also

Filament load is two phase:
1. Load to near proximity of Bondtech gears based on bowden length calibration - now calibration is possible for each filament 1 to 5.
2. Load inside Bondtech gears and disengage MMU gear.

If the first phase doesn't fail, MMU replies ok to the printer, printer initiates second phase - feed circa 15mm of filament and concurrently executes gcode after T<n> code, gcode following T<n> is responsible to load filament into bondtech gears.

AbeFM, 2018-10-04:
> Sadly printer filament sensor is not yet (we are still improving it in
> firmware) so reliable for all types of filament, so we can't rely on it to
> have MMU very reliable. Finda filament sensor in MMU is simpler and more
> reliable for all types of filament, now with bowden length calibration
> separated for each MMU wheel it should be sufficient. We are sorry for not
> catching this problem earlier, we didn't expect production wheels to differ
> in diameter so wildly.

Is averaging something that can be used to improve the reliability of the
optical sensor?

I saw a guy online who used a ball bearing like in the finda, claiming the
sensor does a better job watching the ball.

I changed the loading speed, and the slicer setting where it feeds 13 mm
more filament than it took out. I changed this to 3 with no change outside
of less grinding.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

KarlZeilhofer, 2018-10-10:
@murphy2712 2 days ago a new version 1.0.1 was "released" within Github. I think you must compile it yourself with Arduino, as explained in the README. Prusa hasn't released it yet on their website prusa3d.com/drivers.

Just thinking:   
Could it be, that the actual diameter of the MMU2 gears depends on the type of filament used? I think the teeth of the gears don't dig into the filament exactly the same depth for any filament type. I think this depends on the hardness of the material. Therefore a different average diameter applies to the movement of the filament.

murphy2712, 2018-10-10:
@KarlZeilhofer Right, since my message they added instructions and I was able to compile and flash it.

Several users measured the gears accurately and they indeed have a different diameter. From what I read this fix seems to help a lot. I don't think there are still issues which might be related to the filament type.

AbeFM, 2018-10-10:
It can be, and does happen.

But the issue here is primarily the gear size.

After changing filament settings to do less over-feeding and more speed
matching, I had very little grinding.

 I put on the new firmware, calibrated the filament, and that works great -
but things are less stable, I don't know why. I keep finding the printer
just sitting there in early print, waiting for no discernable reason.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

vertigo235, 2018-10-11:
How does one calibrate each filament with the new firmware?

AbeFM, 2018-10-11:
Like normal, but when filament isn't loaded, left and right buttons select left/right one channel. You have to calibrate each one independently - I suggest counting the steps you have to adjust the first one and start there on the second, third, etc.

vertigo235, 2018-10-11:
Thanks for the response. I was able to stumble my way into it last night. Seems like a great update.