prusa3d/MM-control-01 #33: Please Release a State Description for MMU lights/buttons

AbeFM, 2018-10-02:
A *FULL* descirption of the state of the lights and the MMU would be super useful.

When is a blinking light mean loading? When does the left button unload?

OFTEN I find the MMU in an unusable state - filament sticking out a hole without the selector present, etc.

It can be very difficult to make the MMU do something on purpose, like attempt to load filament, etc. Perhaps it could use some reworking, but just staring and blinking lights it's hard to tell what's even SUPPOSED to be going on.

It is hard to make meaningful suggestions when it's unclear what is happening. I've gotten links to small bits of information that explain how certain procedures work - but never what ALL THREE BUTTONS actually DO in a particular state, how you get in and out of a state, etc.

Something definitive would be a big help!

mkbel, 2018-10-04:
All the information is supposed to be in manual, if something is missing, please comment it there:
https://manual.prusa3d.com/Guide/5.+Preflight+check/759?lang=en
https://manual.prusa3d.com/Guide/Status+LED+indications+explained/762?lang=en
https://www.prusa3d.com/downloads/manual/prusa3d_manual_mmu2_eng_1_00.pdf?2

I will do quick review of manuals and ask responsible persons to add missing info if I found something missing.

closed 2018-10-04 by mkbel

AbeFM, 2018-10-04:
As a start, what are the other four options on the "secret menu"?

I think the MMU needs more functionality, but I'm wary to make suggestions
until I know what it does now.

Thanks for getting this ball rolling!

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

mkbel, 2018-10-10:
Currently all other options in secret menu does nothing. In the next release with bowden length calibration separated for each filament, there will be reset EEPROM. This has no use for ordinary user, our testing department use it for testing if sane defaults are there without bowden length calibration. And to test migration of old calibration to new calibration.

KarlZeilhofer, 2018-10-10:
Collected comments form the code, seems to be the entire functionality
```
//! RG | RG | RG | RG | RG | meaning
//! -- | -- | -- | -- | -- | ------------------------
//! 00 | 00 | 00 | 00 | 0b | Shift register initialized
//! 00 | 00 | 00 | 0b | 00 | uart initialized
//! 00 | 00 | 0b | 00 | 00 | spi initialized
//! 00 | 0b | 00 | 00 | 00 | tmc2130 initialized
//! 0b | 00 | 00 | 00 | 00 | A/D converter initialized
//! b0 | b0 | b0 | b0 | b0 | Error, filament detected, still present
//! 0b | 0b | 0b | 0b | 0b | Error, filament detected, no longer present, continue by right button click
//!
//! @n R - Red LED
//! @n G - Green LED
//! @n 1 - active
//! @n 0 - inactive
//! @n b - blinking

//! @brief Select filament menu
//!
//! Select filament by pushing left and right button, park position can be also selected.
//!
//! button | action
//! ------ | ------
//! left   | select previous filament
//! right  | select next filament
//!
//! LED indication of states
//!
//! RG | RG | RG | RG | RG | meaning
//! -- | -- | -- | -- | -- | ------------------------
//! 01 | 00 | 00 | 00 | 00 | filament 1
//! 00 | 01 | 00 | 00 | 00 | filament 2
//! 00 | 00 | 01 | 00 | 00 | filament 3
//! 00 | 00 | 00 | 01 | 00 | filament 4
//! 00 | 00 | 00 | 00 | 01 | filament 5
//! 00 | 00 | 00 | 00 | bb | park position
//!
//! @n R - Red LED
//! @n G - Green LED
//! @n 1 - active
//! @n 0 - inactive
//! @n b - blinking

//! @brief main loop
//!
//! It is possible to manually select filament and feed it when not printing.
//!
//! button | action
//! ------ | ------
//! middle | feed filament
//!
//! @copydoc manual_extruder_selector()

//!	@brief Show setup menu
//!
//! Items are selected by left and right buttons, activated by middle button.
//!
//! LED indication of states
//!
//! RG | RG | RG | RG | RG | meaning
//! -- | -- | -- | -- | -- | ------------------------
//! 11 | 00 | 00 | 00 | 01 | initial state, no action
//! 11 | 00 | 00 | 01 | 00 | setup bowden length
//! 11 | 00 | 01 | 00 | 00 | erase EEPROM if unlocked
//! 11 | 01 | 00 | 00 | 00 | unlock EEPROM erase
//! 11 | 00 | 00 | 00 | 00 | exit setup menu
//!
//! @n R - Red LED
//! @n G - Green LED
//! @n 1 - active
//! @n 0 - inactive

//!	@brief Select filament for bowden length calibration
//!
//! Filaments are selected by left and right buttons, calibration is activated by middle button.
//! Park position (one behind last filament) can be also selected.
//! Activating calibration in park position exits selector.
//!

//! @brief Set bowden length
//!
//! button | action
//! ------ | ------
//! left   | increase bowden length / feed more filament
//! right  | decrease bowden length / feed less filament
//! middle | store bowden length to EEPROM and exit
//!
//! This state is indicated by following LED pattern:
//!
//! RG | RG | RG | RG | RG
//! -- | -- | -- | -- | --
//! bb | 00 | 00 | 0b | 00
//!
//! @n R - Red LED
//! @n G - Green LED
//! @n 1 - active
//! @n 0 - inactive
//! @n b - blinking

```

mkbel, 2018-10-10:
Sadly this is not complete. I added this documentation only to functions I have been editing. You can get nicer looking output using doxygen.

AbeFM, 2018-10-10:
I can see some holes for sure.

It is a much better starting point than my just wishing, however.

MKBEL - do you feel we should start here, or do you have a more complete base. I'm clearly no match for the folks who write up the manuals at Prusa, but this is close to useful.

It's certainly good enough to start adding suggestions for what functions might help people get successful prints - right now it's all too easy to get in a state where you can't get the printer loaded with the filament you're expecting and turn a print to garbage.

AbeFM, 2018-10-11:
I do not understand why this issue is closed.

LOTS of people are trying to get a *very* poorly functioning piece of hardware working. Watch my videos, and tell me the frustration of trying to figure out what a barely blinking #1 and #4 light are while a steady on red and green on 5 and blinking green 2 means. Like, there's 5 states (off/slow blink/fast blink/dim blink) and 5 channels, and the flashing light inside that only flashes after you hit reset but before it starts.....

I do NOT think the topic of communicating what the buttons do has been handled.

mkbel, 2018-10-12:
> I do not understand why this issue is closed.

The issue is closed, because this issue tracker is related to MMU firmware and not MMU manual. Person maintaining manual doesn't normally read this issue tracker. I forwarded the information about perceived LED blinking documentation being scattered to manual maintainer and asked you to use comment function on https://manual.prusa3d.com/Guide/Status+LED+indications+explained or other Guides. When you comment on  https://manual.prusa3d.com/ , comment reaches directly the person who has rights to update manual.

AbeFM, 2018-10-12:
Thanks! I'll do that, and follow up there.

>

AbeFM, 2018-10-15:
The response that I received on  https://manual.prusa3d.com/  said that this was an issue for GitHUB and that the people there would DEFINITELY be interested in getting the issue fixed.

Perhaps if there WERE a better, documented understanding of what the buttons and lights were supposed to do, there would be less confusion. My feeling on it is there isn't enough documentation for those people to create a definitive list, and it will require the insights of those who know how the device works.

I would really appreciate a single place to document this, and create a factually valid reference. Writing up something pretty could be done one there is valid information to work from.