prusa3d/MM-control-01 #43: MMU2: Flash MK3 MMU board failed

adlamy, 2018-10-09:
Unable to flash the MMU board with firmware 0.9.0 provided together with the last
Einsy firmware 3.4.1
Using Slic3rPE-1.41.1-rc+linux64 and micro USB cable between MMU board and PC.

Installed Einsy firmware 3.4.1 with no problems.

See Slic3r log:

avrdude: Version 6.3-20160220-prusa3d, compiled on Oct 5 2018 at 15:41:54
Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
Copyright (c) 2007-2014 Joerg Wunsch

     System wide configuration file is "/tmp/.mount_Slic3roqlBaQ/usr/bin/resources/avrdude/avrdude.conf"

     Using Port                    : /dev/ttyACM0
     Using Programmer              : avr109
     Overriding Baud Rate          : 57600
     AVR Part                      : ATmega32U4
     Chip Erase delay              : 9000 us
     PAGEL                         : PD7
     BS2                           : PA0
     RESET disposition             : dedicated
     RETRY pulse                   : SCK
     serial program mode           : yes
     parallel program mode         : yes
     Timeout                       : 200
     StabDelay                     : 100
     CmdexeDelay                   : 25
     SyncLoops                     : 32
     ByteDelay                     : 0
     PollIndex                     : 3
     PollValue                     : 0x53
     Memory Detail                 :

                              Block Poll               Page                       Polled
       Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
       ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
       eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
       flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
       lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
       hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
       efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
       lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
       calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
       signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

     Programmer Type : butterfly
     Description     : Atmel AppNote AVR109 Boot Loader
Connecting to programmer: Found programmer: Id = "CATERIN"; type = S
Software Version = 1.0; No Hardware Version given.
Programmer supports auto addr increment.
Programmer supports buffered memory access with buffersize=128 bytes.

Programmer supports the following devices:
Device code: 0x44

avrdude: devcode selected: 0x44
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9587 (probably m32u4)
avrdude: safemode: hfuse reads as D8
avrdude: safemode: efuse reads as CB
avrdude: reading input file "/home/andre/Desktop/prusa3d_fw_MMU2board_0_9_0.hex"
avrdude: writing flash (17286 bytes):

Writing | ##########avrdude: error: programmer did not respond to command: set addr
avrdude: error: programmer did not respond to command: write block

***failed;
####################################### | 100% 0.56s

avrdude: failed to write flash memory, rc=-1

avrdude: safemode: Verify error - unable to read lfuse properly. Programmer may not be reliable.
avrdude: safemode: Verify error - unable to read lfuse properly. Programmer may not be reliable.
avrdude: safemode: Sorry, reading back fuses was unreliable. I have given up and exited programming mode
avrdude: error: programmer did not respond to command: leave prog mode

avrdude done. Thank you.

mkbel, 2018-10-09:
Isn't it known problem with ModemManager? See readme.md.

adlamy, 2018-10-09:
You're right, I deinstalled modemmanager, flash succesfull.
Thanks.

closed 2018-10-09 by adlamy