prusa3d/MM-control-01 #47: Stucked selector, homing keeps failing without any mechanical reason

KarlZeilhofer, 2018-10-10:
My selector got stuck during a 2-filament print, and I couldn't find any reason for that. 

I tried these without any luck:
* removed cutter blade (even though it looked to sit on the right place)
* filed the selector body a bit, since it rubbed a bit on the MMU body
* greased the spindle and the guiding rods (bad English???)
* Power on reset
* MMU reset

What helped:  
* Power Off
* move the selector to the very right end
* Power On

Now the selector is homing correctly again. 

Perhaps this is related to https://github.com/prusa3d/Prusa-Firmware/issues/1181

fnordsh, 2018-10-14:
I noticed that the selector on my MMU2 would not move as easily on the left side. (Most of the time I didn't notice, but especially when homing, it sometimes caused problems.) As it turned out, the steel rods were not 100% parallel. After widening one of the mounting holes, everything works fine, now.
I agree the motion control needs to be improved, but maybe this helps until then. YMMV.

KarlZeilhofer, 2018-10-14:
@fnordsh 
Thanks for your hint. Interestingly if I remove the motor spindle, interestingly the selector is moving a bit easier in the left part. So it is exactly the opposite as in your case.