prusa3d/MM-control-01 #48: home() is called twice on power up on a Prusa i3 MK3

KarlZeilhofer, 2018-10-10:
since this is a lengthy process, it is very annoying. considering how many users are out there, this should really be optimized. 

I assume, the MK3 is sending some command to the MMU to rehome, right? How about homing on that command only?

Greets, Karl

AbeFM, 2018-10-10:
Certainly the MMU could respond with "I just homed, haven't done anything since". Store a "GOOD home" flag?

KarlZeilhofer, 2018-10-10:
@AbeFM good hint! should be doable!

KarlZeilhofer, 2018-10-16:
I found out now, that this isn't done via a command, but by triggering the reset line. 

Possible solutions:
* either the printer doesn't trigger a hardware reset
* or the MMU dosn't home after a reset and waits for an explicit command from the printer to do the homing

AbeFM, 2018-10-16:
Especially since most times the MMU resets, it does so incorrectly. Homing after a successful home most likely means another six or seven resets.

PavelSindler, 2018-10-25:
Thank you for feedback. Printer should trigger hw reset as you need to reset also mmu2, when you press reset button on printer. Homing just on certain command instead of automatic homing on mmu2 reset/power up is good idea.

AbeFM, 2018-10-25:
If you read through the preceding thread, we focused the possibility of a
"reset valid" flag.

Clearly the MMU can't be left in an unknown state. The new customer-led
firmware resets in a third the time and more quietly so it is less of an
issue running that.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes