prusa3d/MM-control-01 #50: MMU Boots without doing anything

AbeFM, 2018-10-10:


KarlZeilhofer, 2018-10-10:
Could it be, that Finda is activated? Is a filament such in the tubes anywhere?

AbeFM, 2018-10-10:
I had the selector off, the ball bearing out, and it was still behaving that way?

This time I reboot it maybe 10 times. The first 9 it didn't move any motors (though I suspect the common shaft moved when "loading", but the drum and selector were utterly unpowered.

The 10th time, it came right back.

I tried booting into "secret menu" mode some of those times, I think the final time it was. Anyway, at a certain point (just as I was about the reflash it) it just started behaving again.

Printer: 3.4.1-1356
MMU 1.0.0-106

KarlZeilhofer, 2018-10-10:
Today I also had a very strange behaviour, as described in #47. But that with the stock 0.9.0 version.

AbeFM, 2018-10-11:
I've flashed 1.0.1-117 and it only homes maybe a third of the time.

Often if will home like normal, but then the selector will stop arbitrarily, maybe at position 3.5 or so. Everything appears to work except for the random positions of all the motors.

Firmware verified ok on upload, it has homed correctly a couple of times, I was nearly able to get through calibrating all the filament lengths. Resetting the printer makes it more likely to come back, but at this point I just want something with a >20% chance of turning on and working.

AbeFM, 2018-10-11:
I guess it's time to dig into the code. The device starts  up, homes the selector, doesn't move the drum, then leaves the selector somewhere between the 3rd and in the home (far right) position, and sits there as if it did its entire boot up sequence.

Any ideas?

AbeFM, 2018-10-11:
I wanted to see if there were any serial port messages to help me debug - the first time I reset the MMU with the serial port plugged in it behaved, and has behaved since.

I've no idea why that is. I didn't even see anything on the serial port. Now even with the cable unplugged, it's working.

AbeFM, 2018-10-11:
That was short lived. I thought 4 reboots in a row was a good sign, told it to print something and it hung again - now won't boot up right anymore.

Maybe there's something in my g-code breaking the MMU? I reset the EEPROM (secret menu/option 2/option3) and got all the lights to flash once, I assume to acknowledge the reset.

Still not blinking the 3rd light on startup, still not homing the drum (which is done later in the code).

AbeFM, 2018-10-11:
I am trying to debug this...

`int8_t tmc2130_init(uint8_t homing)
{
  led_blink(4);//abe
  led_blink(1);//abe

	DDRC |= 0x40;
	DDRD |= 0x80;
	DDRB |= 0x80;

  led_blink(4);//abe
  led_blink(2);//abe
  
	PORTC |= 0x40;
	PORTD |= 0x80;
	PORTB |= 0x80;   

  led_blink(4);//abe
  led_blink(3);//abe

	DDRD |= 0x10;
	DDRB |= 0x10;
	DDRD |= 0x40;
	PORTD &= ~0x10;
	PORTB &= ~0x10;
	PORTD &= ~0x40;

  led_blink(4);//abe
  led_blink(4);//abe

	int8_t ret = 0;
	ret += tmc2130_init_axis(0, homing)?-1:0;
	ret += tmc2130_init_axis(1,homing)?-2:0;
	ret += tmc2130_init_axis(2,homing)?-4:0;
  led_blink(4);//abe
  led_blink(5);//abe

	return ret;
}
`
from tmc2130.c

When I compile, it complains:
sketch/tmc2130.c:260: undefined reference to `led_blink'

for every place I added it. I'm trying to be proactive here, but my printer doesn't work at all and I want to make it work again.

AbeFM, 2018-10-11:
I'm out of my depth here a bit, this is what I wrote. It complies....

`void Abe_led_blink(int _no)
{
  shr16_set_led(1 << 2 * _no);
  delay(40);
  shr16_set_led(0x000);
  delay(20);
  shr16_set_led(1 << 2 * _no);
  delay(40);

  shr16_set_led(0x000);
  delay(10);
}
`

I just added that in right above int8_t tmc2130_init(uint8_t homing)

Now when it boots it just goes 5-4-3-2-1 really quickly, all greenflashes, then sits on 1. I feel this is where a programmer - or someone familiar with the code - might be helpful.

AbeFM, 2018-10-11:
I can define the function and everything works, but when I use it in the init sequence, it acts as described above.

So it's not helping me to debug, all I can do is break it.

If the developers (or kindly others) would tell me how to watch the serial port, or get meaningful feedback, I would love to help. As it is all I can do it keep posting and I don't feel it is helping anyone without some learned interpretations.

KarlZeilhofer, 2018-10-11:
Perhaps there is a bug with the reset/startup sequence with the Trinamic chips...

Do you have some sort of USB to UART adapter? Could also be an Arduino board. When connecting that to the MMU via some jumper wires, you should be able to read debug messages. You could even try to flash a debugging FW onto the printer, then the cabling is fine allready.

AbeFM, 2018-10-11:
...I saw references to messages sent on the UART - but my guess is y ou need to do something to turn on the serial over usb - "debug" flag I set before I compile.

Can someone point me the right way on this?

AbeFM, 2018-10-11:
I should note I am using the printer firmware:
prusa3d_fw_MK3_3_4_1.hex (848,642 bytes)
..which came in prusa3d_fw_3_4_1_mk3.zip

There is also a:
prusa3d_fw_MK3_3_4_1.hex (848,642 bytes)
..which came in prusa3d_fw_3_4_1_mk3MMU2.zip

The files are the same size, but given that my machine has been changing speeds, rebooting randomly, sometimes forgetting the MMU is there altogether with the associated menu items gone..... Could I have the wrong file?

Is there a way to test printer-MMU communication?