prusa3d/MM-control-01 #51: error state when printer boots with filament in FINDA

KarlZeilhofer, 2018-10-10:
As reported there https://github.com/prusa3d/Prusa-Firmware/issues/1240
There should be some procedure to eject the filament, when FINDA is active on bootup. This must be implemented in both firmwares. 

I think it could work like this:
* MMU2 detects filament on bootup in FINDA
* home the idler only
* tell the printer to start and eject procedure
* ask user what to do: (a) heatup and eject (b) manually fix
* if (a): 
  * ask for channel of filament and 
  * ask optional for type of material (for proper temperature)
  * heatup the nozzle
  * do a proper ramming and eject
  * put the filament back into its parking position
* home the selector

jonbet83, 2018-10-10:
Generally if the filament is past the FINDA then something has gone wrong loading or unloading and will need manual intervention, when A print finishes it should I unload and leave the mmu in a condition that allows correct homing etc etc. I see where you coming from but it’s a bit pointless getting the printer to give you options when they won’t work and you have to use manual intervention anyway...

KarlZeilhofer, 2018-10-10:
This could easily happen, when you power off the printer during a print. And for the many reasons what can go wrong, this is very likely to happen. At least in the first times using the MMU2.

mylife4aiurr, 2018-10-10:
I can confirm in newer unreleased mmu2 firmware (i'm on 1.0.1-117) this behavior is fixed. If finda detects filament - selector stays put until you clear the filament. It wont home until filament is not detected by finda, then I hit left button and it homed.

KarlZeilhofer, 2018-10-11:
The problem IMHO is, that the printer doesn't even try to fix the problem itself. Doing it manually is quite annoying.

AbeFM, 2018-10-11:
I agree that if it TRIED to remove the filament, it would be nice.

What is REALLY needed is information in human readable format on the screen, and the ability to make some basic commands: Preload/Load/Unload/Home(selector/drum/all)...

There's a nice screen and button on the printer, and either the MMU needs a screen, or the printer needs to assume some responsibility here.

Perhaps I should go to the newer firmware, 1.00-106 is a lot of headaches, sometimes it boots and just sits there.

mylife4aiurr, 2018-10-11:
I agree the printer should try to clear filament itself during a boot. It could try to retract filament until finda is clear. But You'd want to cap the time retracting uncase filament is stuck and grinding on the pulley.

KarlZeilhofer, 2018-10-16:
and another case: https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f53/psa-eject-filament-before-turning-off-printer--t25185.html