prusa3d/MM-control-01 #53: home_idler() moves also the selector and ignores stall guard

KarlZeilhofer, 2018-10-10:
Further reviewing the code reveals this strange behaviour:
```c
bool home_idler()
{
	int _c = 0;
	int _l = 0;

	for (int c = 1; c > 0; c--)  // not really functional, let's do it rather more times to be sure
	{
		move(0, (c * 5) * -1,0); // <<== WHY moving the selector in a function called 'home_idler()'???
		delay(50);
		for (int i = 0; i < 2000; i++)
		{
			move(1, 0,0);
			delayMicroseconds(100);
			tmc2130_read_sg(0); // <<== Stall Guard is read out, but result is ignored?

			_c++;
			if (i == 1000) { _l++; }
			if (_c > 100) { shr16_set_led(1 << 2 * _l); };
			if (_c > 200) { shr16_set_led(0x000); _c = 0; };
		}
	}
	return true;
}
```

KarlZeilhofer, 2018-10-10:
Note: the comment on the for loop is not from me. In fact it isn't a loop with the given values.

AbeFM, 2018-10-11:
following

mkbel, 2018-11-19:
Partially fixed in https://github.com/prusa3d/MM-control-01/pull/77/commits/0e87ecb0a5c3a33b6d35762e6cba677ee5d623af

closed 2018-11-19 by mkbel