prusa3d/MM-control-01 #55: Check Menu Request before Homing

KarlZeilhofer, 2018-10-10:
Currently the middle button is checked after homing, which takes quite a while. Please let us move this if-statement before calling `home()`. 

```c
	home();
	tmc2130_init(0); // trinamic
	
	// check if to goto the settings menu
	if (buttonClicked() == Btn::middle)
	{
		setupMenu();
	}
```

fnordsh, 2018-10-14:
I would agree, but... the main purpose for that menu is currently to calibrate the bowden tube length(s). For this, you need to have a homed system, anyway. Or am I missing something here?

KarlZeilhofer, 2018-10-14:
of course the system must be homed before entering the menu, but the user should not be enforced to hold the button until everything is homed.

KarlZeilhofer, 2018-10-14:
```c
	bool menuRequest = buttonClicked() == Btn::middle;
	home();
	tmc2130_init(0); // trinamic
	
	// check if to goto the settings menu
	if (menuRequest)
	{
		setupMenu();
	}
```

mkbel, 2019-02-23:
Fixed in MMU firmware, but in order it to has desired effect, printer firmware shouldn't reset MMU if it is responsive.

closed 2019-02-23 by mkbel