prusa3d/MM-control-01 #56: Delays for manual moving Selector necessary?

KarlZeilhofer, 2018-10-10:
What are these (annoying) delays of 500ms good for? Normally an action should be triggered, when a button is pressed, or at least, when it was released.

KarlZeilhofer, 2018-10-10:
```c
void manual_extruder_selector()
{
	shr16_set_led(1 << 2 * (4 - active_extruder));

	if ((Btn::left|Btn::right) & buttonClicked())
	{
		delay(500);

		switch (buttonClicked())
		{
		case Btn::right:
			if (active_extruder < 5)
			{
				select_extruder(active_extruder + 1);
			}
			break;
		case Btn::left:
			if (active_extruder > 0) select_extruder(active_extruder - 1);
			break;

		default:
			break;
		}
		delay(500);
	}
}
```

AbeFM, 2018-10-11:
I'm guessing the later one is to give the extruder selection process time to finish - one could argue this should be done by the select_extruder() function.

The lead in one I don't get at all.

KarlZeilhofer, 2018-10-11:
The Functions are all blocking, so thats no argument.

FaultyLine, 2019-03-05:
debounce function?

AbeFM, 2019-03-05:
I often get the button behaving wrong in a light press. Perhaps the read is
to short to let the circuit settle, but the MMU buttons often act like one
button to the right of the of pressed.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes