prusa3d/MM-control-01 #58: Moving Selector to End to REHOME - Feature Request

AbeFM, 2018-10-11:
Often the issue that I have is the drum losing its home, or the selector.

Moving the selector to the far right (past #5 - with both flashing lights on 5) could/should rehome everything, and have the MMU ask the printer what it's supposed to be doing.

I would suggest re-initializing the stepper drivers as well, here, since oftentimes in this state one or more of the motion components are unresponsive.

Rick-T, 2018-10-20:
+1

I've had the problem a few times aswell and the possibility to rehome would be really nice.

AbeFM, 2018-10-20:
I would very much appreciated some guidance. Is this:

1) A non issue - you could tell me so and close it
2) Something that more information would help
3) Already being fixed

There's no reason for an entry to languish. Let's set it aside, discuss it, or fix it.

mkbel, 2019-02-23:
Probably most common reason for drum loosing home is, that current was switched off by motor driver due to error condition detected. Firmware now detects this condition and re-homes automatically. As side effect of implementing power panic support MMU unit should also handle well reset. So you can reset unit in order to re-home even if filament is loaded in printer.

closed 2019-02-23 by mkbel