prusa3d/MM-control-01 #59: Calibrating Tube Length, Materials

AbeFM, 2018-10-13:
Calibrating the tube length depends on what you're feeding. I can get PLA or PETg to feed and always end up within 1-2 mm on the hobbed gear, but feeding TPU will fall well short (not visible, about 1.5cm short).

I feel that when the printer is printing with a TPU, and knows it, it compensates? I certainly was BARELY getting anything out of the printer, the entire purge was extruding air, then it would pick up as I got to the object.

What I'm wondering is if I need to recalibrate all 5 channels every time I change filament material. If so, it speaks to having a multiplicative constant for each channel, then setting one overall would scale the rest?

KarlZeilhofer, 2018-10-14:
Just like I said. :smile: 

The interface to the printer has something defined to tell the MMU2 the material to be printed. But currently the firmware of MMU2 ignores such values.

clearbucketLabs, 2018-10-15:
usually compression of the filament changes the effective gear ratio making the steps per mm slightly different per type of filament, usually it isn't too much but over x length it would get more pronounced...

AbeFM, 2018-10-15:
A small percentage difference is a big change over the length of the bowden tube - my gears having a tenth of a mm difference was a cm of travel.

The whole feeding of the printer by the MMU is handled pretty badly. Ignored sensors, wrong speeds, wrong lengths, mismatched timings.

It's generally better than standing there yourself.

FaultyLine, 2019-03-05:
I'm having an issue where no matter the tube length calibration I do, the MMU2 pulley forces too much filament into the extruder. I made a mechanical buffer for the filament on the selector that's working pretty great. 

https://www.youtube.com/watch?v=W3ctB_X0odY

AbeFM, 2019-03-05:
Set gear to nozzle distance, extra unload, etc till you get it right. If
the extruder or firmware isn't stock (neither of mine are) you'll need to
make sure they are all set correctly to avoid grinding.

                  Thanks!
                          -Abe.
Sent from my "smart"phone, please excuse brevity and Swype-oes

FaultyLine, 2019-03-05:
It's Prusa firmware - I haven't customized anything and have followed the calibration routine over and over again. I basically gave up on making it work the way it's supposed to and am just waiting for the MMU2S upgrade, which adds the extruder sensor so I don't have to deal with this anymore.

JohnnyDeer, 2021-02-08:
This problematic is solved by MMU2S, I´m closing.

closed 2021-02-08 by JohnnyDeer