prusa3d/MM-control-01 #60: Found a major issue in the code in `void park_idler(bool _unpark)`

KarlZeilhofer, 2018-10-14:
## Current version in 51aea8692a76f39b2071249a87b96afad4bfa1b8
```c
void park_idler(bool _unpark)
{

	if (_unpark) // get idler in contact with filament
	{
		move(idler_parking_steps, 0,0);
		isIdlerParked = false;
	} 
	else // park idler so filament can move freely
	{
		move(idler_parking_steps*-1, 0,0);
		isIdlerParked = true;
	}
	 
}
```

If the caller doesn't check properly, if the pully is engaged or not, the idler gets out of sync and could do the parking or unparking motion twice (!!!). I haven't checked, if all the calls of that function ensure a proper state of `isIdlerParked`, but that's a nogo!

## Suggested code:
First I would rename the function and argument, so we get rid of dumb negations to
`void engage_filament_pully(bool engage)`. Then I would check for the current state of the idler before moving it. 

```c
/**
 * @brief engage_filament_pully
 * Turns the idler drum to engage or disengage the filament pully
 * @param engage
 * If true, pully can drive the filament afterwards
 * if false, idler will be parked, so the filament can move freely
 */
void engage_filament_pully(bool engage)
{
	if (isIdlerParked && engage) // get idler in contact with filament
	{
		move(IDLER_PARKING_STEPS, 0, 0);
		isIdlerParked = false;
	} else if(!isIdlerParked && !engage) // park idler so filament can move freely
	{
		move(IDLER_PARKING_STEPS * -1, 0, 0);
		isIdlerParked = true;
	}
}
```

KarlZeilhofer, 2018-10-14:
`feed_filament()` can be called by the printer interface with a `L`-command, and there the state of the idler is not checked! What could possibly go wrong?

KarlZeilhofer, 2018-10-20:
thanks for that hint!

mkbel, 2019-02-23:
Already fixed.

closed 2019-02-23 by mkbel