prusa3d/MM-control-01 #61: MMU 2.0 cannot recover after powerloss

KarlZeilhofer, 2018-10-16:
this is a link to the original issue in the printer firmware: https://github.com/prusa3d/Prusa-Firmware/issues/1237

From the hardware standpoint of view, it should be possible to recover. 
Could be easily implemented when MMU wasn't moving during power loss. If it was moving, it would be a bit harder to recover. 

But since it's not time critical, what the MMU does, a comprehensive recover procedure should be working fine:
If filament is not in the bondtech gears at the extruder, there should not be any problem at all. 
* home the idler drum only
* engage filament with idler drum
* Retract the filament until the FINDA is clear again
* park the filament on it's default position,
* home the selector too
* aks the printer what filament should be loaded
* ready for printing again

KarlZeilhofer, 2018-10-16:
This is also related to #51

PavelSindler, 2018-10-25:
Thank you for feedback. Do you have working power panic implementation?

KarlZeilhofer, 2018-10-29:
Dear @PavelSindler, I'm sorry, but I didn't go on with the developments, since the missing commitment from Prusa. I cannot proceed with developments, until our branch is merged into the official Git repo due to the comprehensive changes of the code. Otherwise we have to do "back merges" every time the official repo got improvements.

3d-gussner, 2018-10-31:
@PavelSindler and @KarlZeilhofer I opened an issue a while ago https://github.com/prusa3d/Prusa-Firmware/issues/1101 and i think that the MK3 should store next to the MK3 power loss values also the actual mmu2 values like:
- which mmu2 extruder is/was selected (at moment of power loss)
- what status is/was of the mmu2? tool change/load/unload/eject mode

So two additional eeprom power loss values like
- mmu2_active_extruder [active or next 0-5] and
- mmu2_active_extruder_status [1=printing/loaded, 2=tool change not confirmed, 3=load not confirmed, 4=unload not confirmed, 5=eject not confirmed,...]

would be needed to be stored during power loss.

Having this information after power recovery it should be possible to

In case it was printing (mmu2_active_extuder_status=1):
- check if FINDA detects filament
- home the idler drum only
- move the idler relative to "mmu2_active_extruder loaded and printing" position
- send a command to MMU2 that it is power loss recovery mode, which keeps the selector in position and rehomes the mmu2 completely after the unload is finished during the next tool change sequence.
- bring the hotend back to temp
- continue print at line ....

In case of unconfirmed tool change/load/unload/eject command (mmu2_active_extruder_status>1):
- if finda detects filament
  - bring the hotend back to temp
  - home the idler drum only
  - engage the last active used extruder
  - unload filament
  - home the selector
  - recover missed operation from line ... and the print (this should load the right filament)

- if finda does not detect filament
  - home idler
  - engage the last active used extruder
  - load some filament slowly until finda detects it
  - unload the filament to known position or even cut (these 3 extra steps above might prevent that undetected filament stuck in the selector before homing it)
  - home selector
  - bring the hotend back to temp
  - recover missed operation from line ... and the print (this should load the right filament)

I guess there are many many more scenarios that i am missing, like power loss during ramming ...

mkbel, 2018-11-19:
Done in https://github.com/prusa3d/MM-control-01/pull/77

closed 2018-11-19 by mkbel