prusa3d/MM-control-01 #62: Push Buttons for manual select, when filament ejected

KarlZeilhofer, 2018-10-16:
When filament is ejected (and still not removed by the user), the push buttons are still reacting to move the selector, which could lead to a filament-slector-crash. 

Veriosn V1.0.1

PavelSindler, 2018-10-25:
Thank you for your feedback.

mkbel, 2018-11-07:
There is no sensor to detect filament presence in this moment. I have no idea, how this could be possibly fixed.

KarlZeilhofer, 2018-11-07:
I would suggest, that the user has to press the middle button, so the pulley can retract the filament and put it back in place (useful for inspecting the filament tip). If the filament was removed completely, that move wouldn't be an issue anyway. After that, the selector can be released again, to be moved by the left/right buttons.

mkbel, 2019-02-23:
Fixed in 0d5297fb02f42027681789e0fa807d34b887615e and 4bc2f2c5c2d58e7e87416eb17148e56cc9109ee4.

closed 2019-02-23 by mkbel