prusa3d/MM-control-01 #67: MMU2 cant detect no filament @ bondtech gears = layer skips

mylife4aiurr, 2018-10-22:
Mk3/mmu2 only detects filament at mmu2's finda probe. So although filament may be present at the top of the mmu2 and be detected by finda probe - a stringy tip can fail to enter bondtech gears at mk3's extruder. 
Another cause for filament failing to reach bondtech gears is mk3/mmu2 relying on a bowden length calibration that was taken when filament was cut perfectly. However during printing the tips are not perfect, they can be very stringy which throws off the bowden length measurement the mk3/mmu is relying on.

Since the printer has no way to detect or prevent this issue it keeps printing, unaware that filament did not print = layer skips
[image]
[image]
[image]
****
(mmu2 comes in kit form, so it should account for different build tolerances, perhaps a total measure of filament path resistance from spool to bondtech extruder > pass or fail displayed to user

There are dozens if not hundreds of brands of filament. Seems the mmu2 is expecting a level of consistency from all brands of filament rather than being flexible enough to work with all filament. But I've had issuses with even prusa filament and stringing. So rather then expecting perfect tips from every filament - shouldnt the mmu2 actively detect filament making it into the bondtech gears > ask user to fix issue if no filament present)

mylife4aiurr, 2018-10-23:
I realize this issue has been raised in several combined github issues and on the Prusa forums, I'll simply wanted to give the developers a single issue trouble report.

mazkagaz, 2018-10-24:
I am facing the same problem and I may have a solution for the developpers.

I am not sure because my eyes can't look two things at a time but I feel like the printer extruder motor only start when the mmu extruder gear is released. A work arround may be that the mmu extruder motor acts as a bowden extruder during the first 2 cm while the printer extruder starts, and then release the mmu extruder gear only after these 2 cm.

ironjoe13, 2018-10-28:
I also have this problem, and is most certainly repeatable- no matter the color, manufacturer or any other variables of the filament. What I have noted so far is:

Slot 1 skips a layer at 24mm
Slot 2 skips a layer at 18mm
Slot 5 skips a layer at 16mm

As I stated before, I can repeat these results on any model

dancorrigan1, 2018-11-02:
> I also have this problem, and is most certainly repeatable- no matter the color, manufacturer or any other variables of the filament. What I have noted so far is:
>
> Slot 1 skips a layer at 24mm
> Slot [...]

Wow!  Ok.. I thought I was going crazy. What are the odds that I would have layers missing at the EXACT same spot more than once. Sorta good to see someone else having the EXACT same issues as me.

ironjoe13, 2018-12-06:
I have since fixed the issue with a different filament expansion system(loop control), modified selector, selector cover and re-printed the revolver(barrel) with 4 perimeters and 30%

mkbel, 2019-02-23:
This issue should be improved by using MK3S extruder with gear door sensor. And even more improved with https://github.com/prusa3d/Prusa-Firmware/pull/1559, which detects even filament stucking in extruder PTFE tube.

closed 2019-02-23 by mkbel