prusa3d/MM-control-01 #68: Setting the PFTE tube length

mazkagaz, 2018-10-23:
I thought that this had only to be done one time with one filament and then I had the surprise to see that this is depending on the filament number. I don't know why, as the lenght concerned is the lenght on the orange tube, which is the same for all filament but OK : 5 filaments, 5 parameters to tune (isn't it an old code design from MMU1 ?).

Then I tried to tune all these parameters : setting the pfte tube length for the 5 filaments but, when I reset the mmu2, only the last filament i tuned has the corrects length value. So i had to do it 5 times, one per filament.

It would be great to be able to do the 5 at once. (And is only one parameter suffisant ?)

Sorry for my english, i'm french (and tired, it's late).

KarlZeilhofer, 2018-10-23:
The gears have different diameters which get compensated by these parameters.

mazkagaz, 2018-10-24:
So 5 parameters are justified.

Then it would be great to be able to tune these five parameters with only one mmu reset. Actually, on reset, it only save the last length tuned.

koenkooi, 2018-10-28:
I'd love to have an option where it tries to use FINDA + filament sensor first to determine the length and resort to manual if that fails. 
All the filaments I've used so far have been detected by the filament sensor, but I've seen enough reports of shiny/transparent filaments not getting detected.

mazkagaz, 2018-10-28:
> > So 5 parameters are justified.
> > Then it would be great to be able to tune these five parameters with only one mmu reset. Actually, on reset, it only save the last length tuned.
> 
> You are moving the selector to the 6th position and pressing the center button to save the configuration right?

No I am not. According to the manual delivered with the mmu2, page 25-26, the 11th step is to reset the printer. There is nothing about moving to the 6th position. But next time I will ;)

> I'd love to have an option where it tries to use FINDA + filament sensor first to determine the length and resort to manual if that fails.
> All the filaments I've used so far have been detected by the filament sensor, but I've seen enough reports of shiny/transparent filaments not getting detected.
Maybe the simplest way would be to use the mmu sensor only to detect load failure and then use extruder filament sensor to load the filament into the extruder. I mean : on a big distance, the diameters differences between each gears (and filament) affect the total number of mmu extruder motor rotations, but on a short distance between the extruder filament sensor and the extruder itself is short enough (I think) to ignore these differences. And then the mmu wouldn't be so sensitive about filament tolerance : because with a +/-0.05 tolerance, these parameters won't fit an entire big print. I have quantities of these filaments and I think I am not alone in that case.

For example : one lengh parameter to know the approx distance between mmu and extruder to fast load filament near the extruder filament sensor (as it is done at this time). And then, in a second time, use the extruder filament sensor to detect when it's at the right place and a second lengh parameter to load slowly the filament into the extruder gear from the filament sensor to the extruder.

KarlZeilhofer, 2018-10-29:
This would be a very good suggestions, if the laser filament sensor wouldn't detect filament strings, which could easily be 2-3cm in length. And the laser sensor detects them very well. It doesn't seem, that there is any configuration for that sensor, which would only detect a "full filament diameter". 

That's a very, very basic problem with the MMU2, which could be solved only by a better filament sensor setup at the extruder.

mkbel, 2018-11-07:
Book is outdated. Calibration has been changed. Calibration manual is there: https://manual.prusa3d.com/Guide/Service+menu+-+Individual+Filament+Calibration/842

TimothyGold, 2018-11-13:
Ever since receiving the MMU2 I have been really struggling to get a good consistent distance calibrated on all five of my filaments.  I even swapped out the PTFE with another brand trying to get it to be consistent.  No matter what I do I going through the calibration it looks good but after starting a print the idler ends up crashing overtime then gets off and never recovers.  After stopping the print I check the length and find that it is way too long again and again.  Does anyone else have the same problem?

mkbel, 2018-11-13:
Can you record and post video? I am not not sure I can imagine your problem right from textual description.

inquisitiveKangaroo, 2018-11-14:
The original version of mmu firmware had only one bowden length calibration, however, we found out that our subcontractor wasn't able to fit all the gears into a tolerance we demanded so we enabled length cal. for each filament in the following release. The proper bowden length can be influenced by a friction caused during the loading process into the extruder. According to my observations highest influence has bending of the long PTFE tube and entering the extruder body. That's why i highly recommend to make sure top part of your extruder body is properly drilled, especially if you do not use the original printed part. I hope this helps you to get more consistent outcome.

closed 2019-02-23 by mkbel