prusa3d/MM-control-01 #72: Feature Request: Enable MK3 Filament sensor and MMU2 cutter

kcnjamie, 2018-11-05:
Although the MK3 sensor has problems, it works well for a lot of people.  Please enable communication between the MK3 sensor and the MMU2.  This would eliminate the need for PTFE calibrations.  If a user is confident in their sensor, then they can use it. If they are not, then they can disable it and go back to how it is now.

Also, please enable the cutter. this would eliminate the strings and blobs that cause so many issues.

diomark, 2018-11-21:
Not sure about the cutter (seems like it would wear down after x man cuts) - but I think the filament sensor being on is a must..

tedder, 2019-01-16:
filament sensor is enabled in this fork:
https://github.com/TheZeroBeast/MM-control-01

Not an endorsement, just FYI.

bardiir, 2019-02-05:
+1 for auto-calibrating load distance with the Mk3 Sensor

Also when loading material it would be great to watch out if it triggers because if the material gets past the FINDA and not down into the hotend because it gets stuck in the tube then the printer happily continues with the print without noticing despite it having a sensor there that could actually tell that the filament never arrived at the extruder.

As for the cutter... Well you could enable the cutter for the fine strands it pulls only actually. There would be a quite easy move set to do so.

Level the FINDA a bit higher so it does not trigger on debris
Unload the filament until the FINDA does untrigger
Unload to the point where the part that has just stopped triggering the FINDA is right on the cut mark
Cut

This would eliminate the hairs from the previous filament still sticking around the edges and into the tube when loading new filament basically blocking the tube. The hairs are then free to be pushed down the Bowden with the new filament.

mkbel, 2019-02-25:
@kcnjamie Work is in progress. I plan to detect filament load failure https://github.com/prusa3d/Prusa-Firmware/pull/1559 and cut it in such case. https://github.com/prusa3d/MM-control-01/pull/120

JohnnyDeer, 2021-02-08:
First thing solved by MMU2S and second by new FW versions, I´m closing.

closed 2021-02-08 by JohnnyDeer