prusa3d/MM-control-01 #75: Increase maximum permitted length for PTFE length

ruedli, 2018-11-09:
For fitting the MMU on top of a LACK enclosure, the provided orange PTFE is too short. After fitting a longer PTFE tube, I found that calibration limits exist in the firmware.

After changing: "eepromBowdenLenMaximum" (I added 4000 to the existing value) in permanent_storage.cpp

I could calibrate for my longer PTFE tube.

Can this setting indeed be increased for a next release?

awigen, 2018-11-09:
I'd like to upvote this. I also have my MMU mounted on top of the LACK rack and use the following patch:

```@@ -18,11 +18,11 @@ typedef struct
 
 static eeprom_t * const eepromBase = reinterpret_cast<eeprom_t*>(0); //!< First EEPROM address
 static const uint16_t eepromEmpty = 0xffff; //!< EEPROM content when erased
 static const uint16_t eepromLengthCorrectionBase = 7900u; //!< legacy bowden length correction base
-static const uint16_t eepromBowdenLenDefault = 8900u; //!< Default bowden length
-static const uint16_t eepromBowdenLenMinimum = 6900u; //!< Minimum bowden length
-static const uint16_t eepromBowdenLenMaximum = 10900u; //!< Maximum bowden length
+static const uint16_t eepromBowdenLenDefault = 12000u; //!< Default bowden length
+static const uint16_t eepromBowdenLenMinimum = 11000u; //!< Minimum bowden length
+static const uint16_t eepromBowdenLenMaximum = 16000u; //!< Maximum bowden length```
 
Increasing the default and minimum length makes it incompatible with normal length PTFE tubes, but I would like to see the maximum value increased.

ruedli, 2018-11-09:
Thnx for the upvote, were obviously in the same boat here ;-)

In order to avoid problems for users with the existing bowden PTFE, I feel increasing eepromBowdenLenMinimum and eepromBowdenLenDefault is not a good idea. For this reason I vote for **only** increasing eepromBowdenLenMaximum

awigen, 2018-11-09:
Indeed.
> I would like to see the maximum value increased.

ruedli, 2018-11-09:
Good suggestion, I assume the values are there to prevent bouncing of the buttons and still allow a reasonable manner of pressing the button and invoke one step at a time.

What did you find a properly working value was? Alternatively, some kind of acceleration could be implemented, if the same button is activated for more than e.g. 500ms. Will have a look at that, need to review the timer access for this.

ruedli, 2018-11-09:
I suggest acceleration, based upon whether the button was kept pressed or whether the user presses the button one click at a time. This feels very natural.

I can't believe how long it took me to find this simple code (in Buttons.cpp), I at first didn't realize the program falls through that loop all the time :-), Anyhow this solves the acceleration in a nice manner, I assume "pull request" is needed for this suggestion?

		uint32_t last_press=millis();
		uint32_t save_millis=0;
		do
                {
			switch (buttonClicked())
			{
			case Btn::right:
			        if (bowdenLength.decrease()) move(0, 0, -bowdenLength.stepSize);
                                save_millis=millis();
                                if ((save_millis-last_press)>500) delay(400);
                                last_press = save_millis;
                                break;
			case Btn::left:
				if (bowdenLength.increase()) move(0, 0, bowdenLength.stepSize);
                                save_millis=millis();
                                if ((save_millis-last_press)>500) delay(400);
                                last_press = save_millis;
                                break;
			default:
				break;
			}

ruedli, 2018-11-09:
Pull request #76 created for its implementation

closed 2018-11-09 by ruedli