prusa3d/MM-control-01 #78: MMU2 very noisy when idle

dikodahan, 2018-11-16:
I am not sure if this is a specific problem with my unit or if this is by design or just a code fix...
When not printing, just sitting idle, the MMU2 is fairly loud...! It almost sounds like it has a loud fan running though there obviously isn't one. Is this supposed to happen!? Is anyone else noticing this??

franzbischoff, 2018-12-02:
mk25 or mk3?

dikodahan, 2018-12-03:
MK3 (v3.5.0) with MMU2(v1.0.2).
The same issue existed with previous FW on both (v3.4.1 and v1.0.1).

TheHammockGuy, 2018-12-04:
I have noticed this as well.  It appears to be coming from the stepper motors.  
MK3 (v3.5.0 and v3.4.1) with MMU2(v1.0.2).

A friend of mine has also had the same issue and he is using MK3 v3.4.1 and MMU2 v1.0.2

jexoteric, 2018-12-06:
I have this issue on MK2.5. It is incredibly annoying high pitched squealing (with a slight pulsing) that goes anytime the printer is on, idle or running.

johnbiggs, 2018-12-06:
That noise went away for me on 3.5. If I load older firmware it comes back. This suggests that it may be separate from the MMU controller.

TheHammockGuy, 2018-12-06:
The noise did not go away for me upon updating to 3.5.0.  Also, the noise I am hearing is not high pitched, squealing, or pulsing as described by @jexoteric.  This makes me wonder if there is more than one noise issue going on here.

dikodahan, 2018-12-10:
For me the noise did not go away on either 3.4.1/2 or 3.5.0. And as I mentioned originally, for me it almost sounds like an active fan noise.

jexoteric, 2018-12-22:
After updating to 3.5.1 and MMU 1.03 this has gotten worse/more noticeable on my MK2.5.

martysweet, 2019-01-25:
:+1: When idle, the whine from the MMU2 steppers is very noticeable

jexoteric, 2019-02-12:
Something I noticed recently. I had my printer on high power, as it was having trouble homing on auto power. During those times, it was not making noise after a print would finish and it sat idle. I switched it back to auto-power mode, and after that print it was back to having a very noticeable whine/squeal to the MMU2 motors when idle. With the print I just started, the squealing pulse noise can be heard over the printer when on high power mode. I'll keep an eye on it and see if some this behavior is repeatable.

FaultyLine, 2019-03-05:
These issues might be related to the old 1.0.3 and older firmware. I had all the same noise complaints until I moved to 1.0.4-FINAL.

dikodahan, 2019-03-05:
I originally reported this and can confirm that at least for me this issue was resolved with recent MMU firmware.

FaultyLine, 2019-03-05:
Cool! From what I understand, there were a lot of improvements in the firmware for "noise". I know that when I first got my MMU2 the motors would make this sizzling sound when they were doing the initial calibration after a power-on of the printer and now all of that is significantly reduced.

zcubed11, 2019-03-06:
I'm still having this issue with 1.0.5. I'm not sure if the noise is from the steppers or the control board itself but it's extremely annoying if you keep the MMU outside of an enclosure.

chriswal, 2019-05-03:
if you have also flickering leds the noise from the stepper motor is from the 74hc595 register.they have problems with the power (random output on the pins). you need to install buffer capacitors on the power pins of the registers. as some signal lines from the stepper controller are also on the register you get some sound :-) i didnt check if the random signals also have influence to the steps of the motor.
there would also be a possible workaround (not fix) in the firmware. in idle mode the leds gets updated in a loop.
if we get rid of the update the flickering and humming stops in idle mode.

SNDR0, 2019-05-05:
like said from chriswal, soldering a ceramic capacitor 104 between pin 16 and 13 of every 74hc595 you solve the problem

Panayiotis-git, 2019-05-18:
That did the trick for me too.

iamgnat, 2019-06-11:
For me the noise is not the MMU itself, but the power supply and the root cause appears to be the MMU. When idle (even for 24+ hours) the idler motor stay very warm like it is getting power to hold it's current (off all paths) position. Disabling the steppers either via Settings menu or gcode over the serial port has no effect.

When the printer is idle the idler stepper should be deactivated.

iamgnat, 2019-07-03:
Possibly related to #153 (at least it is in my case)

AbeFM, 2019-07-18:
> like said from @chriswal, soldering a ceramic capacitor 104 between pin 16 and 13 of each 74hc595 you solve the problem

Huh, wow, thanks!

markus-i, 2019-10-05:
> 
> 
> like said from @chriswal, soldering a ceramic capacitor 104 between pin 16 and 13 of each 74hc595 you solve the problem

Just coming onto this issue now (while trying to print ASA, so enclosure temperature might be related to that in my case). Why pin 13? That's OE*? Shouldn't that be pins 8 (GND) and 16 (Vcc)?

chriswal, 2019-10-06:
yes it should be GND and Vcc. I think he wrote this because 13 is also connected to GND

Panayiotis-git, 2019-10-06:
And also, it is much easier to solder the caps between pins 13 & 16.

SNDR0, 2019-10-06:
@Panayiotis-git Bingo!

henrikkoch, 2019-10-09:
Is it jitter on the control signals for the stepper motor controllers that are being "killed" by these 10nF capacitors?  And Will this change accomplish silence in the MMU2 unit (and connected motors)?

markus-i, 2019-10-09:
Haven't checked the signals myself, but the proposed capacitors simply stabilize the supply voltage against high frequency distortions. The stepper motors are controlled from three other chips, but some of the control signals to these chips are apparently run through these buffers/shift registers, and occsionally this control information gets lost (or rather misguided) if the shifters don't shift as expected.

henrikkoch, 2019-10-09:
So do you mean. When it is noisy - then it's because some signals gets lost?  The two 8 bit shift registers "only" control  Direction signal and Enable signals for the steppers and the LED's. The Enable signals for each stepper controller, can if noisy, maybe result in some noise from the stepper motor coils I believe. So maybe this is the reason for this will help. In fact just putting the capacitor on U2 should then do the job by its own.

chriswal, 2019-10-09:
Hi, every time the shift register gets updated, there is a spike on the power of the IC. This spike will result in random output of the shift register. the control signals of the motor drivers are on the shift register. so they get randomly enabled / disabled in a very fast time (because the firmware updates the shift register in idle mode / led). this is also the problem of the homing ..the driver gets disabled the motor stops and the sensing detects the endstop.

the capacitor supplies the register with enough power that the spikes are reduced and the IC has normal operating conditions.

henrikkoch, 2019-11-17:
I have placed the 100nF between pin 16 and 13 as well on both U2 and U9. No change at all in the noise

henrikkoch, 2019-11-17:
Changed to ceramic 10nF. Still you can hear noise when idle on the steppers for the MMU2

[image]

henrikkoch, 2019-11-17:
The noise comes from the Revolver selector stepper. Disconnecting P10 on the board for the motor wire the noise disappears!

chriswal, 2019-12-06:
@henrikkoch The capacitors only help against the noise generated by random outputs of the shift registers during update of their pins. it does not help against the standard humm of the steppers when resting between full steps. as the drum or idler needs the most power it is the loudest motor. one solution would be to only use full steps in the firmware of the drum motor.

jurcja1, 2020-02-08:
I want to add my experience.
Had a same issue, when idle and also problem with homing, motor were quite chaotic during movement.

Root cause: cold solder joints.
So I have disassembled board only, detached all plastic connectors, just pull them out, clean all with isopropyl alcohol with brush and let it dry / evaporate. Then I paint board with Flux paste and heat all, locally, by movements of heat gun till Flux paste "evaporate out", try to not heat plastic parts directly (buttons etc) . Be aware, board will be too hot and needs to be cooled down slowly. Then clean it with isopropyl alcohol with brush again, and let it dry. Assembly connectors, put it to printed case and connect to Rambo/einsy. Turn it on, test. 

Of course, this will void your warranty, but solved my problem. This process is typical repair of pcb.

ahmetcemturan, 2020-05-08:
Can anyone explain why the complication with the shift registers was made?  I understand that the board was developed based on the Arduino Leonardo.  It seems to have enough pins to drive 3 stepperse without the Shift Registers,  Get the input for the buttons and probe.  Wouldn't the steppers then be silent?    Also another workaround would possibly be incorporating a planetary gear for the Drum (Revolver).  It would get rid of the need to keep the idler motor enabled all the time as Planetary gears only work in one direction.  Also we could the Use a small pancake motor.

chriswal, 2020-05-08:
the shift registers are used for the leds. also the shift register needs io pins for data input.
if the capacitors are in place they are working fine.

yes if you are using only the outputs of the arduino the stepper would be "silent" (not affected by the io register problem) but you have no leds.

there are a lot of workarounds possible:
the obvious one ... install the capacitors
then you can do it via software ..dont update the leds when moving the motors. dont update leds in idle mode
in the case you dont want the motor to be on all the time:  home and position the drum motor before its used (takes more time for switching filament) also you need an endstop for the drum (else the homing sequence is loud).