prusa3d/MM-control-01 #90: Mismatched behavior when canceling prints

jexoteric, 2018-12-13:
The behavior of canceling a print from the LCD (when printing from and SD card) acts as expected: the print stops, and then the printer unloads the filament from the hot end back to MMU. When a cancel print command is recieved from Octoprint (likely from other USB sources as well, but not confirmed) it stops the print but fails to unload the filament back to the MMU, causing a jam.

PavelSindler, 2018-12-17:
What "cancel print" command do you mean? 
I think that adding "M702 C" (unload current filament) to octoprint "cancel" gcode scripts should solve this issue.

jexoteric, 2018-12-17:
The only option for canceling a print when printing with Octoprint is via the web interface. The LCD menu on the printer does not have a cancel option unless printing from SD (at least in my observations).

I will add M702 to my Octoprint scripts and see if that helps.

jexoteric, 2018-12-19:
Adding "M702 C" to octoprint does indeed fix this, and is a good work around, but it doesn't address the issue directly in why does the printer handle canceling a print on the LCD and from a USB connection differently?

closed 2019-02-25 by mkbel