prusa3d/MM-control-01 #94: selector doesn't home properly on mk2.5

6ill9re9, 2018-12-18:
Sometime it does sometime it doesn't...
Most of the time it goes to the left and force and then home wrongly.
Revert to official which home properly but somehow the loading was better with the beta, less grinding .

jwveldhuis, 2019-02-03:
o, it's not just me then. With 1.0.3 fw it does not home at all now, back to 1.0.2 it most of the times homes, but sometimes not.. Very annoying..
@6ill9re9 did you found out how to resolve it in the meantime?

6ill9re9, 2019-02-08:
1.0.4-rc1 --> worse, still using 1.0.2

jwveldhuis, 2019-02-08:
Looks like they forgot MK2.5 runs at 12V instead of 24V so StallGuard does not function (properly)?
Very annoying! Even 1.0.2 is not homing in a predictable way :(

6ill9re9, 2019-02-08:
I also have mmu2 on mk3 and it also doesn’t work...

Grismo, 2019-02-08:
Just installed 1.0.4-rc1. Wanted to test it together with new MK2.5 firmware 3.5.2-rc1. Failed already during initialization. The bug described in issue [#1422](https://github.com/prusa3d/Prusa-Firmware/issues/1422) has changed now a bit. Selector is moving to the right again like in firmware version 1.0.2 (which is much less noisy) but now the selector is off by 2 mm to the other side (towards the stepper). The gap is 3 mm now! Loading any filament is not possible!!! 
[image]

KDanielK, 2019-02-10:
I have the same issue with 1.04-rc1, the gap is huge and I have to manually screw it to the left to make the loading work. I also have the MK2.5

Grismo, 2019-02-11:
Although I tried to fix the problem for me I ask myself why just a few report such a problem and for others it works without problems. I cross checked FW 1.0.2 and found that it uses the same values for steps after homing, and it works fine. Very strange. It's like the distance between homing position and position for extruder one has changed.
The best solution would be a possibility to adjust a misalignment manually in the setup menu were bowden length for every extruder is defined. That value should be stored permanently.  In that way one could adjust misalignment if necessary.

KDanielK, 2019-02-12:
I never had this issue until the latest firmware, it might be due to the new homing function. Earler it homed really hard to the left, now it homes to the right and only very sligthly. The difference is huge, earlier it was hammering to the left, now its going rly slow and gently to the right and to the left stopping way ahead of the walls.

Grismo, 2019-02-16:
Seems that this problem is solved now. #116 solves the problem. I just tested it and it works fine.

FaultyLine, 2019-02-19:
I have an MK2.5 with MMU2 running 1.0.4-RC1 and have the problem described in this bug.

jwveldhuis, 2019-02-22:
MK2.5 @ 3.5.2 and MMU2 @ 1.0.4 has solved this issue for me.

FaultyLine, 2019-02-23:
Just upgraded to 1.0.4-FINAL and all homing issues are gone for my MK2.5. Thanks everyone.

closed 2019-02-25 by mkbel