prusa3d/MM-control-01 #97: Idler Motor Disabled?

sarf2k4, 2018-12-20:
I started another print, which is the same with my previous issue, but this time I forgot to align f2 because the print uses f1 for 4 hours, I couldn't wait that. printer asked for attention, which I did, but the f2 not fully loaded into the nozzle. I immediately pause the print, eject filament, then load filament to nozzle. At this point, I'm unable to load the filament through mmu2 anymore because the idler motor was disabled and wont engage no matter what, while mmu drive motor not moving, as well as the lcd panel stuck at "loading filament x...." while mmu2 not doing anything to load. Even if I were to manually put filament through selector to trigger finda active high, it was no use. Another failed print gone to waste.

I am using MMU2 1.0.3

PavelSindler, 2018-12-20:
Thank you for reporting this. We have observed this issue on some test printers also. Seems like mechanical issue. Try to check all cables, especially 24V power supply cables for mmu2. We have new firmware with detection that this happened and reinitialization/rehoming of axes, but it is not fully finished yet.

sarf2k4, 2018-12-21:
Thank you for the reply and solution. However, I observed closely on the idler part, during loading of f2, it just got shifted and wont catch the filament properly. I then opened up the idler cover to see what is going on while the unit is turned on, the idler able to move freely up to a certain angle but when reaching that max angle, I'm forcing the motor to skip the step already. Thought that the idler drum is busted, but I decided to tighten the screw that's gotten loose. No wond

closed 2018-12-21 by sarf2k4

FaultyLine, 2019-03-05:
Have you tried any of the newer firmware versions? I'm currently using 1.0.4-FINAL (even though a new version has replaced it) and am having great lick with it so far. For my 1.0.3 was defective (a lot of others complained too).

sarf2k4, 2019-03-05:
I'm on the latest mm-control firmware, but I'm unable to test this as I don't have the resource to print multi material

AbeFM, 2019-07-18:
This seems trivial to test, no? Load one filament, pause, reach in and try to rotate the drum. I've certainly had the drum get lost and seen filament spit out the wrong hole with the latest firmware.