prusa3d/MM-control-01 #103: Improvements  to homing, movement and stall detection

Kolous, 2019-01-06:
Now when the secret is out that the MMU2 firmware is crap and the community needs to fork it to implement basic functionality like proper homing itself: [implemented in this fork](https://github.com/TheZeroBeast/MM-control-01/releases) it is time to get all the functionality to official firmware.

The whole topic was discussed already in this thread: [this thread](https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f53/improvements-to-the-mmu-2-0-controller-firmware-t25189.html) and [this is where we want to get](https://www.youtube.com/watch?v=ZncfiwjcNes)

Current version (1.0.3) can't even home properly and is pushing against left wall. The same for 1.0.2 but at least in 50% of cases it aligns against the holes otherwise it is 1-2mm off....

mkbel, 2019-02-25:
Do you experience this with MK3 24 V or MK2.5 12 V printer?

Kolous, 2019-02-25:
MK2.5 12V

mkbel, 2019-02-26:
Problem with homing 1-2mm off with MK2.5 was fixed in v1.0.4. Pushing against left wall (end stop detection too sensitive) affecting some of the units wasn't fixed yet.

Kolous, 2019-02-27:
The two problems are connected. When pushing against left wall than is always 1-2mm of. And when homed properly to the right the alignment is correct. Can't imagine to fix one without other. I have now to restart the MMU unit so twice to get proper homing and best result is when I move to filament 4 or 5 before restart.

mkbel, 2019-02-27:
Can you try this release, if it solves your problem? https://github.com/mkbel/MM-control-01/releases/tag/v1.0.6-HOME

Kolous, 2019-02-27:
Got error during firmware uploading (using Linux Slic3r AppImage) and now the unit doesn't light up nor show up port on PC. Any idea how to recover it?

mkbel, 2019-02-28:
Slic3r should ask you to push reset button on the unit if it is not found after you push flash button. Pushing reset button leads to bootloader activation. Reset button is located next to USB connector.

mkbel, 2019-02-28:
[image]

mkbel, 2019-02-28:
Board is not powered from USB, you need to power up whole printer. Wait some time after powering up the printer, as printer resets MMU before starting communication with MMU, this could potentially lead to update interruption. You can tell printer - MMU communication has finished by F<nr.> of F? shown in the middle of printer status screen.

mkbel, 2019-02-28:
Yeah sorry, if the MMU is not communicating - which is probably your case F<nr.> of F? will never be shown, so just wait several seconds after you see printer status screen.

mkbel, 2019-02-28:
If Slic3r is not asking you for pressing MMU reset button, try later version of Slic3er.

mkbel, 2019-02-28:
If this is not working, you can check, if bootloader is running after pressing reset button. There are two SMD LEDs on MMU printer circuit board not visible when enclosed in plastic cover. LEDs are marked RX and TX. When the bootloader is activated by pressing reset button, it is active for 5 seconds. When it is active, one of  those LEDs (red) is changing brightness (breathing animation).

mkbel, 2019-03-05:
@Kolous Any progress there?

Kolous, 2019-03-05:
Just recovered the MMU unit and upgraded it to 1.0.5. There is a problem in the linux Slic3r FW uploader failing with different error messages. On a Win7 machine the FW was uploaded on first spot. Will test the 1.0.5 now.

mkbel, 2019-03-05:
On Linux there can be problem if you have modemmanager package installed. ModemManager tries to open new CDC devices discovered and is blocking such devices for about 60 seconds. But the bootloader is active for only 5 seconds, so Slic3r has no chance to open device in time. I have removed modemmanager package on my machine, as I don't need it anyway. If you need it, there exist some fix without removing this package.

mkbel, 2019-03-05:
@Kolous Can you try this release, if it solves your homing problem? https://github.com/mkbel/MM-control-01/releases/tag/v1.0.6-HOME

Kolous, 2019-03-10:
No need, so far 1.0.5 improved the homing significantly. So far no pushing against left wall and the offset is correct. The last think to be improved is pushing of idler body.

mkbel, 2019-03-11:
Thank you for your reply. I am closing this.

closed 2019-03-11 by mkbel