prusa3d/MM-control-01 #108: Feture Request: Control Feed Speed

ch98-1, 2019-02-02:
Is it possible to create a way to change the speed that the filament is fed and unloaded through the PTFE tube to extruder? I have a filament that grinds unless its fed through very slowly (cheap BVOH from push plastic) and was wondering if it was possible to switch speeds without flashing the firmware every time.

FaultyLine, 2019-02-19:
+1 this whole request. The Prusament Black Galaxy I have seems to have difficulty unloading/loading during a MMU Benchy I'm trying to print, and because the MMU2 pulley body grinds away at the filament, I end up with a missing chunk that causes it to never get to the extruder - then I get missed layers.

AbeFM, 2019-07-18:
My MMU only grinds when something is wrong in the extruder. Filament doesn't get released on unload (often due to a string), or hits one of the multiple transitions above the extruder gears, or DOES load, but printer doesn't notice and when it tried to feed high speed into the nozzle it'll chew up the filament.

Most of the issues are logical, I believe. Keep the teeth clean, too, and play with the spring tension. . :-)