prusa3d/MM-control-01 #112: 1.0.4 RC1 Increased unload speed - too loud!

cosmith71, 2019-02-08:
I'm not a fan of the increased unload speed in the new RC.  It is loud and screechy versus 1.0.3.  I can hear it from much farther away and will definitely be a problem at night.

Consider slowing it back down to reduce noise?

jlmyra, 2019-02-09:
I like the increased unload speed. It works well!

cosmith71, 2019-02-09:
I should clarify.  I _like_ the speed, but I can see where the noise could be problematic.

Perhaps a stealth mode?

Grismo, 2019-02-12:
I don't hear any difference on MK2.5. It is as loud as before, but maybe that's due to lower voltage?

mkbel, 2019-02-23:
Higher speed applies only for 24 volt printers, so MK2.5 is not affected. We plan to add silent mode.

cosmith71, 2019-02-23:
> Higher speed applies only for 24 volt printers, so MK2.5 is not affected. We plan to add silent mode.

Awesome!  Thanks!