prusa3d/MM-control-01 #113: Recovering from "Drive error and counting"

jlmyra, 2019-02-09:
Drive error correction and counting- 
1.)"the MMU2 unit stops printing and starts flashing continuously with red and green LEDs"- Do you mean all 5 LED's flash red/green? )(I encountered this on an unload fail overnight)
2.)How does one recover a print if this error state occurs? 
3.) You state, "A possible cause can be broken MMU2 power supply wiring, connectors etc" How does one diagnose and determine the cause of the Trinamic drivers not  providing enough power for steppers?
Thanks!

PavelSindler, 2019-02-15:
1. yes all 5
2. printer should automaticly recover (rehome axis) during print and then continue printing. If problem is repeating on the same movement, printer give up recovery attempts (after three consequent power fails) and stays in this error state (blinking all 5 LEDs red and green). You should then check your hw and you can then try to recover by pressing reset button on mmu unit.
3. It is very individual. We had some test printer with bad connector (never figured out what actually was wrong there) and when we moved with cables during stepper movement, power went off. That is why we decided to add this feature.
4. Basicaly when all 5 LEDs are blinking red and green, power fail counter increments.

There has been fix in this function since 1.0.4-RC1. Some power failures which occured right after powering printer on could be false. Wait for stable release.

domesticatedviking, 2020-07-02:
Pressing the reset button on the MMU2 installed on a MK2.5 results in a reset of both the MMU and the printer.  This "all 5 lights" blinking behaviour is an unrecoverable error on the MK2.5.   Please remedy.