prusa3d/MM-control-01 #117: Selector misaligned with MMU2 Pulley body

FaultyLine, 2019-02-19:
I've upgraded to 1.0.4-RC1 from 1.0.2 (1.0.3 is incapable of homing correctly...) and immediately noticed that the selector is misaligned with all filament holes. Appears to be about 1 mm to the right, instead of centered on the filament pathways. My MMU2 is installed on a MK2.5. Downgraded back to 1.0.2. I'm happy to provide more details, or pictures/video if it's helpful.

FaultyLine, 2019-02-19:
Another user has posted of the same issue I'm having. 
[https://shop.prusa3d.com/forum/general-discussion-announcements-and-releases-f53/firmware-1-0-4-rc1-for-the-mmu2-unit-t27618-s10.html](url)

Grismo, 2019-02-19:
That is the same issue as reported in issue #94. Meanwhile devs have released a hotfix for that (see #116). That one works well and solves the issue.

FaultyLine, 2019-02-19:
That's wonderful news. Please excuse the ignorance, but how does one go about applying the hotfix? Do I need to get the code and compile it myself? Also - I'll look a bit deeper before I post another bug report that may have already been posted. I looked but didn't see it. I'm new to GitHub - thanks for baring with me :/.

Grismo, 2019-02-19:
I compiled it myself. It's fairly simple if you follow the [instructions](https://github.com/prusa3d/MM-control-01#windows) of the windows section here in the repository. Download the source code of the repository in one folder. Download the MM-build-env-Win64-.zip and unpack it in a second folder. Make a third folder for the compiled firmware. Run the configure.bat in the second folder. And then run build.bat in the third folder. That's it.

mkbel, 2019-02-23:
Fixed in  v1.0.4.

closed 2019-02-23 by mkbel