prusa3d/MM-control-01 #122: Idler can't find position or won't move with both 1.0.3 and 1.0.4

JeffAGreen, 2019-02-24:
I keep having to reflash back to 1.0.2 to get my MMU to get it to work properly with the firmware. With 1.0.3 the idler doesn't move at all. With 1.0.4 it doesn't seem to be able to find the correct positions and it doesn't restart properly when the machine reboots, or when the MMU reboots. I tried erasing the EPROM, as well. As soon as I go back to 1.0.2 it seems to work correctly.

FaultyLine, 2019-02-25:
Are you running a MK3 or MK2.5 printer?

JeffAGreen, 2019-02-25:
Sorry. That would be useful to know. I’m running and MK3. With the 3.5.3 firmware. Just so you know, I went back to 1.0.2 and calibrated the MMU and it’s working just fine again. I’m assuming there are features in 1.0.4 that are worth having, but I can’t seem to get my MMU to accept the more recent firmwares.

Thanks for any help you can offer.

JeffAGreen, 2019-02-25:
Well, I spoke too soon. Apparently my MMU doesn’t want to work with any version of the firmware now. I would get a sporadic issue where the MMU would crash/jam when printing and when I tried to restart it after clearing the issue, it would go to the far right position and the light would indicate it thinks it’s at the filament 1 position. Often, the only way to fix this is to turn the computer off and turn it back on. Obviously, that kills whatever print I’m trying to finish. Sometimes, when it does go back to position 1, it also decides to extrude just enough filament from filament 1 to jam the print again when it tries to move into position, even if the print isn’t using filament 1.

Here’s a video of the reset not working, if that helps…

https://www.dropbox.com/s/dl2jqso4ib0dxos/Video%20Feb%2024%2C%2011%2056%2035%20PM.mov?dl=0

FaultyLine, 2019-02-25:
Oh wow - that's ugly. I suppose you've already verified that the selector moves freely (if the motor is removed)? Also, looks like MK3 3.5.3 firmware is for the MK3S. If you have an MK3 (w/o the new extruder design that has the opto-mechanical filament sensor that makes it an MK3S) then you might want to try MK3 3.5.2 + MMU 1.0.4 firmware again to see if your issues go away.

JeffAGreen, 2019-02-25:
> @JeffAGreen <https://github.com/JeffAGreen>  There is no 3.5.3 firmware for MK3 printer. Haven't you flash MK3S firmware to MK3 printer by accident? This would explain extreme extruding during filament load, as there is no IR door sensor to stop the feed when filament arrives bondtech gears. Also bug in MK3S 3.5.3 is forcing MMU into silent mode, probably this has less power and could explain that selector stops moving after homing if there is more mechanical resistance. Can you rotate trapezoidal screw easily by hand when the unit is not powered? There can be some dirt in trapezoidal nut, or the nut thread can be damaged. If the nut thread is damaged, I would recommend you to ask you our support for new nut, as it is nearly impossible to remove remaining plastic burrs. I stabbed my fingers several times with needle fixing it.

Yes, what’s the sign for “red in the face emoji” for making a silly firmware upgrade? I am now running 3.5.2 with 1.0.4 and the older combo seems more predictable. 

 

But there’s still some odd behavior after restart. Sometimes after the restart, I hit a button to move the loader the next filament position and it goes through the full range of movement again before it moves to the next filament position – it’s like it resets again. 

Also, does 1.0.4 add any way to get to the calibration menu other than holding the button down during a reset? It seems like there’s something odd happening when I try to enter calibration mode. Like it also does that second full movement step…

JeffAGreen, 2019-02-28:
So, I tried 1.0.4 with 3.5.2 and it’s still a disaster for me. It gets lost all the time. It doesn’t restart properly. It partially loads to filament 2 on every restart, and then feeds in some filament for no good reason, no matter how many times I try to calibrate it. And sometimes it pretends to restart to position 1, but when I hit the button to advance it, it restarts again and goes to position 2 – and usually feeds out a little filament again. It crashes significantly more for me than with 1.0.2. I’m very frustrated. I have gone back to 1.0.2.

mkbel, 2019-02-28:
> But there’s still some odd behavior after restart. Sometimes after the restart, I hit a button to move the loader the next filament position and it goes through the full range of movement again before it moves to the next filament position – it’s like it resets again.

This change is intentional. Selector is not homed after restart, but it is homed just before it needs to be moved. So in your case, hitting right button starts moving selector to the second position, but it is homed first. This new behaviour is there to support power panic and also allows you to unload filament from printer menu, if filament is already loaded to nozzle if you turn it on.

mkbel, 2019-02-28:
> So, I tried 1.0.4 with 3.5.2 and it’s still a disaster for me. It gets lost all the time. It partially loads to filament 2 on every restart, and then feeds in some filament for no good reason, no matter how many times I try to calibrate it.

@JeffAGreen Can you post video of this problem?

JeffAGreen, 2019-02-28:
It will probably be next week. I switched back to 1.0.2 so that I could get some prints done and I’m gone this weekend. 1.0.2 easily finished the print, at least. 

Jeff Green
@Jeffagreen
Mobile: 248-310-9015

JeffAGreen, 2019-02-28:
Ah, ok. That would be fine, then, if it didn’t also load a little filament from Filament 2. Also, it seems to sometimes NOT pick up the command at the start of the print and it doesn’t home, and doesn’t load the correct filament. It seems like I need to push a button to get it to home just so that I can then start a print from the SD card. I will try to capture the behavior on video when I can load it again. I can see that part of this is that I’m confused about what the new behavior is for the MMU with this firmware.

AbeFM, 2019-07-18:
Did you make any progress on this?
I had issues with cabling, all sorts of odd behaviors, typically one motor or another wouldn't move or would lose its place partway through a print. Reseting, etc, reflashing didn't fix, but wiggling the cables often would.