prusa3d/MM-control-01 #124: Loud Startup Sequence

sarf2k4, 2019-03-02:
I often print at night but having mmu2 plugged in, it is so noisy during its startup sequence, it will do twice idler alignment then another homing sequence when moving the selector or loading filament via lcd menu.

I had a case where the screws to fasten the idler to the motor gotten loose due to the vibration during the startup sequence. I don't know what driver that mmu2 board uses, I'm assuming that it uses the same tmc2130 (I think) as the selector motor as well.

My suggestion is that to reduce the idler motor crashing/homing duration and thus reduce risks of loosen screws as well as reduce loud noises

FaultyLine, 2019-03-05:
If the duration of the clicking was reduced, the idler might not make it back to the home position. EX: If the idler and selector are in filament 5 position and you reboot the printer, the clicking is only for a very short period of time. If it's in the filament 1 position and you reboot, the clicking lasts for the entire time the idler *would* take to move into the F1 position *if* it were in the F5 position. I'm not sure that makes a lot of sense there, but I'm trying to explain it in a way that does... 

Basically, if the idler moved for a shorter period of time, and it was in the furthest position from "home" it wouldn't make it home in a lot of cases. 

I have no idea how hard it would be to implement, but I don't understand why the stall detection isn't being used for the idler assembly, preventing the clicking noise all together? I suspect getting the stall detection going would solve the issue you are bringing up here.

sarf2k4, 2019-03-05:
I suspect as much that the reply would be about the idler not having enough
time to reset its position.

It is true that im asking about stall detection for the idler motor to
prevent those crashing noises. X and y uses it, z axis uses it when it is
not leveled, selector uses it as well.

chriswal, 2019-06-01:
Hi,
the stall detection is not implemented on the idler , because it doesnt work with the current settings.
to get stall detection working with the 2130 you need to speed up the axis. also it is tricky because it depends on the user (screws/springs tension). On each filament position you get a high load to the idler motor. so if it stalls you cant really be sure its in the home position. The homing sequence should be ...move in the opposite direction until stall... move in the home position until stall and count the steps..
if your in the right range your home ;-)

sarf2k4, 2019-06-02:
I am sure that this can be implemented by fine tuning the sensitivities. Last year we had issues of the mk3's random false positive crash detection during prints and now that problem has been addressed by adjusting its sensitivities for stall guard detection.

Even the selector motor may have triggered the stall detection during the startup sequence when the selector is at F1 position before going all the way to the right side. 

I am sure that there will be a function that will validates that if the axis for selector and idler has homed or not, which is by checking the finda probe if its triggered or not first.

jltx1, 2019-06-11:
@sarf2k4 I agree with you about the horrific cacophony the MMU2 makes to wake up my family.  It ruins the MK3.  I stopped using it altogether.
@chriswal I assume there are problems with trying to speed up the idler axis sufficiently to utilize stallguard?  I understand the load increases as it passes each filament, but the bearing should smooth that some.  The endstop is a hard limit.  Surprised those two events can't be distinguished.  Won't your proposed sequence have the same issue?  I'd be ok with it if it just hit once versus slamming a dozen times loudly.   Reading the firmware, it tries to move 2000 steps even if it is already homed, with stall guard disabled.  So annoying.

chriswal, 2019-06-11:
the sequence i proposed makes only sense if stallguard is working. its working on the selector. i made a lot of tests with stallguard also with different speeds . yes you get it to work. but you cant get it to work reliable on different machines. and perhaps it changes if the temperature of the motor changes too much.
here is a video of my test version (21 second video ) 
https://photos.app.goo.gl/VGtwNQ3AxHh5Q1zW8
the other one is a fix for the hc525 problem and has also the loud homing.
perhaps we should include a command that the user can tune the sg parameter for the idler in stealth mode

chriswal, 2019-06-11:
also ther is a bug in the firmware that if you once loaded a filament it never unloads in the eeprom and you get the idler homing on power on and then again with the normal homing.

jltx1, 2019-06-11:
Ok, I just saw your fork and pull request.  Nice!  video looks promising.  I will give yours a test.  Let me know if I can help with coding or testing.  I'm not familiar with hc525, will search...

jltx1, 2019-06-11:
Ah, that explains the double homing.  Thx!

chriswal, 2019-06-11:
my pull request has the same loud startup sequence .....sorry 74hc595 thats a shift register and it used for the leds and the enable and direction signals for the stepper. and if you have flickering leds in idle mode ...or a humming motor there is a power problem with this ICs (random outputs on change) .. they need buffer capacitors on the power pins of the ICs. 
my fix is only for the people who cant install this. it doesnt make updates on the leds while moving the motor in homing.

jltx1, 2019-06-11:
What stall guard value worked for you with your machine before you disabled it?  I will try it just to see.   BTW, what is polarity of sg?  Does higher mean takes more load to trigger?  comments I found in code are confusing.

chriswal, 2019-06-11:
sg=0 highest load (stall) i used 0 to disable it you can use 1
if it doesnt work try changing
#define TMC2130_SG_THR_2       10
in config.h
lower value means more sensitive... so if it crashes lower the value if it stalls on the gears set it higher

you can also set the trials on the selector to 1
	int _c = 0;
  shr16_set_led(2 << 2 * 2);
	for (int c = 5; c > 0; c--)   // not really functional, let's do it rather more times to be sure
	{
		move(0, c * -18, 0);

the for ( int c =1    in stepper.cpp in selector homing

chriswal, 2019-06-11:
you have to test it with filaments in each input, because the load is higher.

chriswal, 2019-06-11:
you should change stepper.cpp
to this     
if (_idler > 0 && _acc <2) {sg_idler= tmc2130_read_sg(AX_IDL);}
    if (_selector > 0 && _acc <2) {sg_selector= tmc2130_read_sg(AX_SEL);}
    if (_pulley > 0 && _acc <2) {sg_pulley= tmc2130_read_sg(AX_PUL);}

_acc<34 to <2 then the acceleration phase is over

chriswal, 2019-06-11:
this is the video with the software in the pull request and the mentioned changes.
https://photos.app.goo.gl/icbQXJ3NB7PxKTZs9

jltx1, 2019-06-11:
I tried move_with_stallguard(2000, 0,0,100); and it works even with filament in all channels!  Nice work!  will play more to see if reliable or need to lower.
What does the TMC2130_SG_THR_2 do?  I see it written to a register in tmc2130, but when does that kick in?  Your software routine:        
 if((sg_selector <= _sg || sg_idler <= _sg || sg_pulley <= _sg) && _sg > 0){break;}  
does what I thought the tmc was supposed to do, i.e. trigger when threshold crossed.  OR do interrupts need to be enabled?  not familiar with this chipset.

chriswal, 2019-06-11:
thats the threshold for the stallguard . the chip has a register where you can read the value of the load on the motor ..0 is the highest load without load it should be 1023 i think. with more load it will be lower.
which value you get depends on the threshold if its lower its more sensitiv.

no only value read from a register

jltx1, 2019-06-11:
Hmm, sounds more like a (negative) gain control than a threshold.  reading app sheet for tmc2130 now.

jltx1, 2019-06-11:
BTW, I turned off cool step.

chriswal, 2019-06-11:
ok ..what changed?

jltx1, 2019-06-11:
I just turned it off for debug since data sheet said to tune SGT first.  I've turned it back on and still works fine.  So, only changes I made to your routine, other than 100 sg for home_idler, is in move_with_stallguard:
if (_idler > 0) { idler_step_pin_reset(); _idler--;  delayMicroseconds(800); }  // increase velocity for stall detection, was 1000
I was trying to get closer to 50 RPM (see fig 14.2).  This is still slow, maybe 25 RPM, but should help.
Note that I did NOT implement your changes to home_selector function, just home_idler.
Two more change I like that you can try are:
in move_proportional: if (delay > 600 && _selector > _start) { delay -= 10; }  // increase change speed, was 900
in move: 	if (_selector > 0) { selector_step_pin_reset(); _selector--;  delayMicroseconds(500); } // increase homing speed, was 800
let me know if it works for you.

chriswal, 2019-06-12:
i tried also different speeds .down to  delayMicroseconds(100);
but i didnt recognize much difference.
the changes i made are mostly for the problem if you have flickering leds and homing problems with the selector, and shouldnt change the normal behavior
if you dont suffer from this ..it doesnt matter. you could change the original code. stallguard is partly implemented there also.

jltx1, 2019-06-12:
@sarf2k4 in addition to above, comment out from setup() in main: //        motion_set_idler(filament);
it appears completely superfluous and saves an unnecessary homing on startup/reset.  Because s_selector_homed = false; you will always get a proper homing anyway to get things positioned correctly again.  The one in setup is not needed and just adds noise.

jltx1, 2019-06-12:
@chriswal another data point: sg very sensitive to motor temp.  They recently increased holding current for some reason.  This causes idler to get quite warm over a long time.  Warm works with the 100 setting I used, but when I let cool it is too low and idler clicks.  A higher setting would work better when cold.  If there was a temp reading available, could adjust sg level until it warmed up.  But will have to live with fixed value.

chriswal, 2019-06-12:
@jltx1  the motion_set_idler(filament) is used for error handling ....if you have loaded filament and not unloaded it.. and you turn of power--you loose the home position of idler and selector ....you cannot move the selector.. so they home the idler and move it to the last loaded filament that you can unload it.
there is a bug in the firmware that the loaded filament is not removed in eeprom when unloaded .. its an easy fix ...there is a commit in my pull request for this.... fix it in the code...load and unload filament.. and the initial idler homing is gone...

Panayiotis-git, 2019-06-12:
@jltx1  This is also my finding. And this was the problem with The zero beast's version. That is the reason why I'm trying to figure out a a gearbox solution (3.5x copied from skele). Yesterday, I tried for first time, but there are some more changes needed for the steps and speeds. It was the most consequent till now.
The gearbox will also increase the torque, so the current could be lowered.

jltx1, 2019-06-12:
@chriswal So that is there for power recovery.  Too bad.  Was hoping to remove it.  I saw your fix and included it.  
@Panayiotis-git Interesting idea.  I'll follow that development.

chriswal, 2019-06-12:
what about a simple end switch on the idler?

jltx1, 2019-06-12:
that would be a better answer, but are there any spare pins left, and are they ported out somewhere accessible?

chriswal, 2019-06-12:
It looks like D7 on the SPI Plug is free. Thats the one plug that is free on the board.

chriswal, 2019-06-12:
also it would be possible to do a mod on the finda pins.

darBis, 2019-06-12:
> @chriswal 
> 
> my pull request has the same loud startup sequence .....sorry 74hc595 thats a shift register and it used for the leds and the enable and direction signals for the stepper. and if you have flickering leds in idle mode ...or a humming motor there is a power problem with this ICs (random outputs on change) .. they need buffer capacitors on the power pins of the ICs.
> my fix is only for the people who cant install this. it doesnt make updates on the leds while moving the motor in homing.

Many thanks for pointing out to missing capacitors. I have had issues with selector homing at power on and flickering leds. And since learning that those shift resisters are also used for stepper control it all makes sense now. I have never imagined they would be missing on the board. :-( I was not expecting that from Prusa. After I had slapped 100nF cap, as per datasheet, at each IC it is all working great now.

Regarding homing of the idler, I am also considering dedicated homing switch, possibly using Prusa's new optical sensor for extruder. Current homing procedure is quite crude solution.

chriswal, 2019-06-14:
i will try also to make a version with a unified parking position for all filaments when idle. that is exact the homing amout away. so when you unload the filament it parks at the homing distance .. and if you are printing (filament loaded) in the normal position. then you have only loud homing if you have a power outage ..or you left the filament loaded after printing

AbeFM, 2019-07-18:
> @chriswal another data point: sg very sensitive to motor temp. They recently increased holding current for some reason. This causes idler to get quite warm over a long time. Warm works with the 100 setting I used, but when I let cool it is too low and idler clicks when hitting endstop. A higher setting would work better when cold. If there was a temp reading available, could adjust sg level until it warmed up. But will have to live with fixed value.

It would be cool if you could use the SG value, say, during homing, to calibrate. Every time you change Z (or feed N-th filament), home the motor, read the threshold (to get an idea of this temperature effect)....

>Yesterday, I tried for first time, but there are some more changes needed for the steps and speeds..... The gearbox will also increase the torque, so the current could be lowered.

Wouldn't lower torque make stall guard LESS sensitive?

Panayiotis-git, 2019-07-19:
As noted earlier the calibration of the SG is very sensitive to several parameters. Back when I was using the TZB's firmware, I had managed to calibrate it so the SG was mostly working. You see, as the idler is currently designed, it allows for a very small window of working SG and my configuration will not work for everybody.
The use of the gearbox widens this window. I've tested it for a while and it worked even if I haven't fine-tune the configuration. But it requires some changes in the stepping of the motors. Especially hard, due to limitations on time availability and my capabilities, is the move_proportional method that rotates simultaneously the idler and the selector. I haven't managed to make it work smoothly as I would like using a gear_ratio parameter. It mostly rotates the one and then the other.

AbeFM, 2019-07-22:
Ah, you're saying the minimum torque to move reliably and the stall torque are so close it is unreliable. That makes sense!

1) Moving both at once to save time? Why do they need to be synchronized? You can't start both processes, and when they both finish move on, time still saved?

2) Given that you just want to increase the stall force for homing, could you swap in a bigger motor instead of a geared one? It's $20, but it's not crazy.

Thanks once again for sharing your understandings, Mr. P.

chriswal, 2019-08-02:
i have finished the endstop version for the idler.
https://www.thingiverse.com/thing:3783694
https://github.com/chriswal/MM-control-01/tree/Pinda_Endswitch

if there is no endstop/or its not working the firmware behaves like before (loud homing sequence)