prusa3d/MM-control-01 #147: Perceived Code Problem post troubleshooting

that1guyuknow, 2019-05-19:
Hi, during a print I have run into the issue that after troubleshooting either a loading or unloading issue on the mmu2s that the mmu2s trues to load the filament before the nozzle is heated up which can lead to problems. For me that meant over extrusion and knocking over a wipe tower that was 30 hours into a print. I may be misunderstanding this problem, but I think it is worth looking into and seems like a simple code fix.

mkbel, 2019-06-05:
Sadly this can not be easily fixed. It sounds like you have had some failure of filament loading in wipe tower. You resolved it by loading to nozzle from printer menu, and then resumed the print - and the wipe tower gcode tried to load more leading to over extrusion in wipe tower. I will think about the problem, but the solution is non-trivial. I will check instructions in manual - if we have some - and try to update it.

mkbel, 2019-06-05:
I checked the user manual, and for MMU Load failed, there is no recommendation, that you should use load to nozzle from printer menu. If you didn't use load to nozzle from printer menu, just resumed the print, it woldn't over extrude in wipe tower. It would load filament itself without over extruding material.

closed 2019-06-05 by mkbel