prusa3d/MM-control-01 #151: Feature Request: Smart PTFE tube length calibration

iamgnat, 2019-06-11:
Setup:
- MK3S+MMU2S
- 5 auto-rewind spools rather than a buffer.
- 3.7.0/1.0.5

Given that it appears that on the 3S the MMU is now using the IR sensor to help it with loading, I swapped in a 1m PTFE tube between the MMU and extruder to see how it would handle it. I'm happy to say that it worked with no, but it could use improvements.

My test was 5 20x20x0.2mm squares with each assigned a different filament. I first ran it with the normal tube and it worked as expected. I then ran with the 1m tube and it worked as well, but with a substantially longer run time (30m vs the 13m with the normal tube).

The issue is that while loading starts normally, after it runs the "fast load" portion it starts "searching" for the IR sensor. It does this by advancing the filament a bit at a time until the IR sensor trips. How long that takes is clearly going to be dependent on the PTFE tube. After finding the IR sensor it continues on as normal. When it comes time to unload for the next tool change it does the same thing in reverse looking for the FINDA. The whole process repeats for each load and unload.

The way I would like to see it work is that it would go through this slow process on the first load, store the information, and then reuse it for all future load/unload procedures for the current print.

This would allow for great flexibility for the end user as to where the locate their MMU in relation to the extruder with a minimal impact to the printing time.

iamgnat, 2019-06-11:
So I had a poke at the code for a few minutes and I see that the load/unload are controlled here:

- https://github.com/prusa3d/MM-control-01/blob/master/MM-control-01/motion.cpp#L146﻿
- https://github.com/prusa3d/MM-control-01/blob/master/MM-control-01/motion.cpp#L115

What interests me most is that as I dig into the `BowdenLength::get()` function in [permanent_storage.cpp](https://github.com/prusa3d/MM-control-01/blob/master/MM-control-01/permanent_storage.cpp#L84) is that there appears to already be logic for dynamically determined PTFE lengths. At least in a 10 minute pass I don't see the `increase()`, `decrease()`, and `~BowdenLength()` methods being used though.

As `eepromBowdenLen` is an array yet the MMU2 has only one PTFE from the unit to the extruder, my guess is that this is legacy code from the MMU1 that never got cleaned out.

Hopefully I'll get a chance to test it, but my assumption is that code can be added to the `unload_to_finda()` and `motion_feed_to_bondtech()` functions to utilize the `increase()`, `decrease()`, and `~BowdenLength()` methods to store the new size if more than N `do_pulley_step()` calls are required.

I haven't dug far enough, but I am assuming that `active_extruder` is still changing and each tool would then be populating it's own value in `eepromBowdenLen`, so a change I would make to `~BowdenLength()` is to have it store the value to all `eepromBowdenLen` entries. This would mean that once the initial load is done for the purge line at the start of the print it shouldn't have to do the work during any other tool change.

AbeFM, 2019-07-18:
Totally with you - I applaud that the printer 'figures it out', but it's a good assumption the tube won't change length significantly from the first layer to the second, and a lot of time could be saved.

I find the 1st load fails like 70% of the time, but the rest of the print is generally ok. It's in the logic, I'm sure, but I rarely get answers to genuine questions on here. :-\