prusa3d/MM-control-01 #154: MMU2 flashing - no response

grzlus, 2019-07-01:
I have a problem with my MMU2. When I try to upgrade the firmware 'm getting such error. I tried different USB cables (I do not have different computer to use).

* **Device**: Prusa i3 MK3 + MMU2
* **OS**: MacOS Catalina
* **Software**: Slicer PE 1.41 or PrusaSlicer 2.0
* **Firmware**: 
  - Prusa i3 MK3 - TheZeroBest 2.1.8 or original 3.7.2
  - MMU2 - TheZeroBest 2.1.8

```
Flashing Original Prusa MMU 2.0 Control, looking for VID/PID 0x2c99/3 or 0x2c99/4 ...
Found VID/PID 0x2c99/4 at `/dev/cu.usbmodem145401`, rebooting the device ...
Found VID/PID 0x2c99/3 at `/dev/cu.usbmodem145401`, flashing ...
avrdude-slic3r -v -p atmega32u4 -c avr109 -P /dev/cu.usbmodem145401 -b 57600 -D -U flash:w:0:/Users/grzlus/Downloads/MM-control-01.hex:i

avrdude-slic3r: Version 6.3-20160220-prusa3d, compiled on May 20 2019 at 09:53:15
                Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
                Copyright (c) 2007-2014 Joerg Wunsch

                Using Port                    : /dev/cu.usbmodem145401
                Using Programmer              : avr109
                Overriding Baud Rate          : 57600
                AVR Part                      : ATmega32U4
                Chip Erase delay              : 9000 us
                PAGEL                         : PD7
                BS2                           : PA0
                RESET disposition             : dedicated
                RETRY pulse                   : SCK
                serial program mode           : yes
                parallel program mode         : yes
                Timeout                       : 200
                StabDelay                     : 100
                CmdexeDelay                   : 25
                SyncLoops                     : 32
                ByteDelay                     : 0
                PollIndex                     : 3
                PollValue                     : 0x53
                Memory Detail                 :

                                         Block Poll               Page                       Polled
                  Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
                  ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
                  eeprom        65    20     4    0 no       1024    4      0  9000  9000 0x00 0x00
                  flash         65     6   128    0 yes     32768  128    256  4500  4500 0x00 0x00
                  lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
                  calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00
                  signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00

                Programmer Type : butterfly
                Description     : Atmel AppNote AVR109 Boot Loader

Connecting to programmer: avrdude-slic3r: butterfly_recv(): programmer is not responding
avrdude-slic3r: ser_recv(): read error: Device not configured
avrdude-slic3r: butterfly_recv(): programmer is not responding
avrdude-slic3r: ser_drain(): read error: Device not configured
avrdude-slic3r: ser_send(): write error: Device not configured
avrdude-slic3r: ser_recv(): read error: Device not configured
avrdude-slic3r: butterfly_recv(): programmer is not responding
avrdude-slic3r: ser_send(): write error: Device not configured
avrdude-slic3r: ser_recv(): read error: Device not configured
avrdude-slic3r: butterfly_recv(): programmer is not responding
avrdude-slic3r: ser_recv(): read error: Device not configured
avrdude-slic3r: butterfly_recv(): programmer is not responding
avrdude-slic3r: ser_send(): write error: Device not configured
avrdude-slic3r: ser_recv(): read error: Device not configured
avrdude-slic3r: butterfly_recv(): programmer is not responding
Found programmer: Id = ""; type =     Software Version = Hardware Version = avrdude-slic3r: ser_send(): write error: Device not configured
avrdude-slic3r: ser_recv(): read error: Device not configured
avrdude-slic3r: butterfly_recv(): programmer is not responding
avrdude-slic3r: ser_send(): write error: Device not configured
avrdude-slic3r: ser_recv(): read error: Device not configured
avrdude-slic3r: butterfly_recv(): programmer is not responding
avrdude-slic3r: error: buffered memory access not supported. Maybe it isn't
a butterfly/AVR109 but a AVR910 device?
avrdude-slic3r: initialization failed, rc=-1
                Double check connections and try again, or use -F to override
                this check.

avrdude-slic3r: ser_send(): write error: Device not configured
avrdude-slic3r: ser_recv(): read error: Device not configured
avrdude-slic3r: butterfly_recv(): programmer is not responding
avrdude-slic3r: error: programmer did not respond to command: leave prog mode
avrdude-slic3r: ser_send(): write error: Device not configured
avrdude-slic3r: ser_recv(): read error: Device not configured
avrdude-slic3r: butterfly_recv(): programmer is not responding
avrdude-slic3r: error: programmer did not respond to command: exit bootloader
avrdude-slic3r: ser_close(): can't reset attributes for device: Device not configured

avrdude-slic3r done.  Thank you.
```

jettoblack, 2019-07-07:
Hi @grzlus did you manage to get around this problem? I am also coming from TZB firmware to MMU2S, and I'm having the same problem. The MMU2 shows up as a USB COM port, but as soon as I try to flash the firmware, it disconnects. I reset it and it reconnects, but again, every time I try to flash the firmware it immediately disconnects. Any suggestions? Thanks.

grzlus, 2019-07-07:
@jettoblack so far I was able to flash it on Windows. Now I'm updating the hotend to Mosquito, so I don't have a chance to try it again on MacOS.

AbeFM, 2019-07-18:
I can get it to go from Arduino - make sure you don't have Slic3r running in the background, it takes the port so you have to kill the program.

Also, the MMU2 only enables the USB port for programming (unless your updater does things right) for the first few seconds, so I hit "send" and just before it starts looking, I hit reset (see my "reset button" on thingiverse, you pretty much need one to use the device), which makes the correct port show back up.

grzlus, 2019-07-19:
@AbeFM from Windows it worked like a charm. I did not get this special 2-3 seconds gap you're describing. Just plugged it, opened Slicer, updated and it is working now.

pascalscheffers, 2019-07-21:
I had the same issue on MacOS Catalina (beta). Resolved by using windows.

AbeFM, 2019-07-22:
> @AbeFM from Windows it worked like a charm. I did not get this special 2-3 seconds gap you're describing. Just plugged it, opened Slicer, updated and it is working now.

Quite right, I see this issues when uploading from Arduino IDE. But it can be frustrating when you don't know why it is happening.

paukstelis, 2019-07-27:
Same issue under Linux. I don't have Windows so this sort of kills using PrusaSlicer for MMU2 firmware uploads.

BastR, 2019-08-02:
Same here.

Is it possible that this is specific to macOs Catalina and the end of 32bit support??

urbaniak, 2019-08-25:
Just got the same issue with MMU flashing, but MK3s works fine, so shouldn't be related to 32 bits drop in Catalina

saeugetier, 2019-09-22:
I had this problem on Ubuntu as well. Found the answer in this [stackoverflow post](https://stackoverflow.com/questions/40951728/avrdude-ser-open-cant-open-device-dev-ttyacm0-device-or-resource-busy). 

Uninstalling "modemmanager" helped.

Thierry-Myron, 2019-10-26:
Same issue, MMU2S won't upgrade using MAC OS Catalina.
However, MK3S will upgrade without issues (3.8.1.2869.hex)

orohde, 2019-11-03:
Same issue, MMU2 not updating when using macOS Catalina, Printer upgrade working fine.

jeffwalter, 2019-11-11:
I used my work laptop that is still on Mojave and the MMU2S upgrade went fine. My personal laptop on Catalina (10.15.0) failed in the exact way as @grzlus saw. I chatted with Alessandro at Prusa and he's going to make sure the dev team knows this is (still) an issue.

Yaroon, 2019-11-17:
Just chiming in to say the same. I finished assembling the MMU2S for my MK3S and the firmware won't load onto it.

 I'm on macOS 10.15.1 Catalina and get:

Connecting to programmer: avrdude-slic3r: butterfly_recv(): programmer is not responding

anton-matosov, 2019-11-18:
I am having exactly the same issue on macOS Catalina 10.15.1
My MMU2S is at version 1.0.5, latest is 1.0.6, so I probably can just give up for now... and wait until next time

anton-matosov, 2019-11-18:
This seems to be related https://github.com/arduino/Arduino/issues/9290

anton-matosov, 2019-11-18:
Glanced quickly through discussion. Some report it is fixed in hourly builds of Arduion, others in 1.8.10, others suggest Apple has to fix USB susbsytem. Also there seesm to be new Catalina compatible bootloader for atmega32u4 https://github.com/adafruit/Adafruit_AVRProg/commit/add3cbd174dc56a739c6e76da9ec4ef23b73f064

smurfix, 2019-12-08:
Interesting, I guess getting the new boot loader onto the MMU would require an actual programming adapter (and soldering a 2x3 header onto the MMU board – cheaper than buying a Pogo pin header).