prusa3d/MM-control-01 #155: v1.05 startup sequence doesn't match manual

bynnek, 2019-07-16:
The manual and factory reset instructions say the selector is supposed to move on the first bootup.  on v1.05, it does not.  It basically doesn't do anything until you hit a button or the printer tries to do something.  At that point it homes the selector.  This makes it hard to know when to hold the right button combo to get into the service menu.

boraca, 2019-07-18:
Are you sure you connected your steppers correctly? Selector homing should occur at bootup.

AbeFM, 2019-07-18:
Mine doesn't, and I read somewhere (PR stuff, official) that it doesn't home the selector on purpose. There's some logic which I forget.

I wonder if anyone at PR has used the MMU? They could comment if any of them have.

Panayiotis-git, 2019-07-19:
I think it depends on whether you have a filament loaded during a shutdown. Not sure though.

AbeFM, 2019-07-22:
I just recently stumbled across this (and lost the source again), but it is certainly on purpose. Probably the manual should be updated, if that's the case.

A bad honest answer is better than a false one or nothing at all.

My NIH grant came in, so I ran a 14 month study:
Case 1: Turn off printer without filament in it, selector @ position 1, then turn on printer (still without filament in it)
Drum homes (slow), printer resets MMU, drum homes (fast) again.
Press selector button, drum homes a 3rd time, selector homes, then moves to position 2.

Case 2:
Turn off printer without filament in it, selector @ position 1
Add filament into MMU sensor then turn on printer (now with filament in it)
Drum homes (slow), printer resets MMU, drum homes (fast) again.
Press selector button, drum homes a 3rd time, all lights flash
No additional movement

(remove filament, lights go green, press selector button, homes, and moves as before.
Move selector back to position 1.

Case 3: Put filament into sensor. Turn off printer with filament in it, selector @ position 1
Turn on printer (with filament in it)
Drum homes (slow), printer resets MMU, drum homes (slow) again.
Press selector button, drum homes a 3rd time, all lights flash
No additional movement

It seems the same to me, excepting the frequency of the banging during the second homing. That may be random every time, but I see no difference between having filament in on when you shut it off. There would have to be a function that wrote that to EEPROM on shut down, like power panic. It seems unlikely and I don't know why you would even care?

JakoobCZ, 2019-07-29:
> The manual and factory reset instructions say the selector is supposed to move on the first bootup...

Hi, can you please point me to the manual, which states this?
Thank you

bynnek, 2019-07-29:
It's the PDF included in the latest firmware release here: https://www.prusa3d.com/drivers/

I've noticed there is a relatively new guide here that has the correct startup sequence: https://help.prusa3d.com/article/rgbbt7x68p-mmu-2-s-factory-reset

chriswal, 2019-08-02:
with V 1.05 no homing at power on. except for the idler if you have once loaded a filament to the printer.
due to a firmware bug this will never be deleted and the idler homes and goes forward to the last filament loaded.
if you press a button or send commands from the printer where the MMU2 should do anything the state will be checked, and if its not homed, its homing the idler and the selector. 
example 
selector is in position 3 you press button left (nothing happens because on startup it thinks its in position 1 so --left no function 
if you press button right it homes the go to position 2
if you press the middle button . it homes stays on position 1 load filament

they removed the homing on startup because on init it worked before that way:

the MMU2 homed
the printer send a reset and the MMU2 homed again

as the MMU2 is loud on homing they wanted to fix that, but with the bug they got it back when you first loaded a filament (because the idler is the loudest part)