prusa3d/MM-control-01 #156: Keep track of PTFE length, don't feed more than XX% longer than tube.

AbeFM, 2019-07-18:
If my printer misses a feed (selector lost position, drum lost position can both cause filament to miss PTFE entirely, malfunctioning IR sensor can feed feet of filament into nozzle and make a huge blob on bed), it would be nice if the amount the printer tries to feed were less than ~1.2x as far as the average of the last 5 loads.

chriswal, 2019-08-04:
i think the problem is that they are in load mode if filament is not feeded (finda sensor triggered)
and load mode is infinite or at least very long. without sensors to detect if filament is loaded you would have to take care that all filaments are loaded before starting a print, then if the finda sensor is not triggered after some time this is an error. the mmu2 only feeds the tube length and then some length for feeding into extruder. if there is a big blob i think the printer sends commands to the mmu2 for feeding again. so this need to be fixed in the printer firmware

TheZeroBeast, 2019-09-11:
Haha, annoying hey @AbeFM