prusa3d/MM-control-01 #158: MMU2 sometimes skipping steps on homing (1.0.6)

vintagepc, 2019-08-19:
I've noticed that sometimes the selector will not home normally. Often it will do the "tap-tap-tap" routine but recently (possibly related to stealth mode changes?) it's started running into the end stop for longer periods until it detects the home; instead of a quick tap one or more will be a longer "BRRRRRRRRRRRRRRT" where it is skipping steps. 

Screws on the leadscrew nut and motor are tight; I'm going to try switching between stealth and normal mode to see if that seems to be related.

chriswal, 2019-08-21:
hi this is a problem of the hc595 buffer. you need to add buffer capacitors or try my firmware fix.

https://github.com/chriswal/MM-control-01/archive/Pinda_Endswitch.zip
this firmware is the same as 1.0.6 i added fixes and the possibility to use an endswitch on the idler.
https://www.thingiverse.com/thing:3783694

vintagepc, 2019-08-23:
Just to note that running in normal mode it still does this. So it is not related to the stealth mode changes in 1.0.6