prusa3d/MM-control-01 #159: 1.0.6 MMU2S Unit Resets On Filament Change

gummerch, 2019-09-05:
Setup:

- MK3S w/Bondtech on 3.7.2
- MMU2S on 1.0.6
- Skinnydip post-processing script (only way I can get the tips serviceable)

Every 10-100 tool changes the MMU unit resets when a filament unload is initiated. Process is:

- Extruder moves to wipe tower and initiates unload as expected
- Unload sequence proceeds all the way through the ejection from the extruder. This is the point 
   when the MMU is supposed to take over.
- MMU unit acts as though I've pressed the reset button on the side. Lights cycle and selector
  barrel homes.
- By the time the MMU has come back online the filament is stuck in the extruder as the printer 
  does not detect an issue with the MMU and is still trying to feed filament, expecting the next to 
  be loaded.

Any insight would be greatly appreciated.

workinghard, 2019-09-10:
Most likely you're encountering [same issue like me](https://github.com/prusa3d/Prusa-Firmware/issues/2086). I've tried to describe what's happening there and the fix as-well.
The change in the pull request solved this issue for me but unfortunately there is no response from the devs on it. Forced to use my custom build FW because of that :-/

AbeFM, 2019-09-14:
Mine reset randomly as well - I didn't note what was happening as carefully, I just ran alternate firmware again.

mwjones1971, 2019-09-29:
@gummerch I am having the same issue, and was almost to the point of requesting an RMA replacement of the MMU board, but now that I've seen others having the exact same problem(s) I suspect it is something in the firmware.

I've seen this issue in the 1.0.5 and 1.0.6 firmware on the MMU with 3.7.2 and 3.8.0 on the Mk3S, but I also have seen since installing 1.0.6 where the MMU will stop with all 5 sets of LED's flashing, and the only way to recover is to reset the MMU (recessed button on side), although one of those resulted in the wrong color being loaded/printed, which it got back in sync with the right color after the next unload.

At this point it is hard to leave the printer unattended for any length of time because of the frequent shutdowns (and wasting a lot a filament clearing the ground down and jammed sections)

vertigo235, 2019-10-05:
I keep getting this red/green 5 lights flashing too.

chriswal, 2020-07-29:
problem description
https://github.com/prusa3d/MM-control-01/pull/167