prusa3d/MM-control-01 #162: Preheat Options on Print Pause

AbeFM, 2019-09-14:
When Pause Print is selected, the printer sets the nozzle temperature to 0.

Most of the time Pause is selected, it is to work on something. Since the head lifts, I suggest there is no harm in leaving the nozzle heated for ~1-2 minutes, or until there are no user interactions for a short period.

Also:
I'm printing at, for instance, 220/50. I have a plethora of options to preheat to, but as none of them are the temps I'm using, I have to wait for a temperature change, load, unpause, then wait for it to go to the print temperature from the preset one.
If "Temperature When Paused" was an option it would be a LOT more useful.

At minimum, it is inconvenient. For some materials, it would ruin a print.

AbeFM, 2019-11-18:
I've used this implemented now on 3rd party firmware, everyone who has used it really liked it.

workinghard, 2019-11-18:
This function is being handled by the MK3s firmware, not MMU.

AbeFM, 2019-11-18:
Good point, I'll post it over there.

closed 2019-11-18 by AbeFM