prusa3d/MM-control-01 #163: Show MMU2 sensor info on main lcd all the time

smitty97, 2019-10-15:
Digging through the long menu to get to Support... Sensor info is not very helpful when fiddling with MMU loading problems.  In some cases, the menu is not even available onscreen to select!  Then the MMU does its repeating filament load movement attempts and fails and the user is left wondering why.

If, on the main lcd, there were some persistent numbers for the FINDA & IR values (even a 00 in the bottom corner) it would help so much.  Bonus: Make the digit inverted colors when it's at an undesired value. (i.e, when loading to the IR it will be inverted '0' until filament arrives, then changes to normal text '1'. When unloading, it will be inverted '1' and then become normal '0' as the filament leaves)

These digits should remain especially at times when the lcd is not showing the main menu (and unable to navigate to main) with text such as "mmu needs attention" and "loading filament".

When it gives up and gets to the "mmu load failed" screen, it would be nice if the indicator digit blinked, showing what sensor failed to provide the expected result.

Mjankor, 2019-11-25:
Yes please!

rdpeake, 2020-10-24:
Was looking into this recently for asking on more error output. From what i can tell, the filament loading code, and serial decode all happens on the main loop, so whilst the MMU is doing an operation it is not responding to other requests from the printer which would mean they couldn't query it for its state as it stands.