prusa3d/MM-control-01 #165: Mintemp error while waiting for user fix kills MMU print

bretcobrageek, 2019-10-29:
I have a MK3S (3.8 firmware) and MMU2S (1.0.6) and print from octoprint.  It has started to get cold at nights, but still like to open a small window to have good ventilation in the printer room.  But I have discovered that if the MMU has an issue that needs the user to fix something it will park the extruder and turn off the heater which makes sense.

Unfortunately, this means the hot end will reach room temperature which, with the window opened and being below freezing outside, can be under the 15C mintemp limit. Once the mintemp error occurs, the printer must be reset, effectively killing a very long print that could've easily been finished with a simple MMU fix and restart.  

Is it possible to disable the mintemp error check on the hot end when you know that the heater is off?  Or maybe reduce the trigger point to 5C instead of 15C?  (15C is really not that cold for a room temperature).

workinghard, 2019-10-29:
The mintemp error is coming from the MK3s firmware, not MMU firmware. I suggest you close this ticket and open a new one there.

bretcobrageek, 2019-10-29:
Unfortunately, there have been several issues filed there regarding this and all have been rejected.  I will try with the MMU complication and see if that helps convince someone to address the issue.  

Opened this issue:  https://github.com/prusa3d/Prusa-Firmware/issues/2299 . 
Will close this one.

closed 2019-10-29 by bretcobrageek