prusa3d/MM-control-01 #170: stepper stalling during load and unload

bynnek, 2020-02-15:
Issue: I have this issue intermittently but the stepper motor that spins the bondtech gears for loading and unloading seems to be stalling.  The behavior is as follows:  the filament starts to get pushed through the feed tube as the speed ramps up.  As it gets near the load / unload max speed, the stepper makes a loud buzzing sound and the filament stops feeding.  After a few seconds making that noise and being stalled, the extruder continues to slowly push filament in small increments till it gets to the extruder.  It happens on all five of the filament selector locations.

Printer: MK2.5s MMU2s
Firmware: 3.8.1 @ 1.0.6

What I've tried to resolve:
- Reflash both printer and MMU2s firmware / performed factory reset of both
- Cleaned out everything in the MMU to ensure there is no debris stuck in it
- Checked filament feedpaths to ensure filaments can move with little resistance through the entire system
- Unplugged and replugged all the connectors in the MMU2s 
- Double checked the power wire connections to ensure tight

Anyways, I'm at a bit of a loss.  It doesn't always happen, but when it does, it'll happen dozens of times in a row.

chriswal, 2020-07-29:
is it possible that the load is too high in the teflon tubes? i changed mine to tubes with 3mm inner width because of the load the 2mm version produced at that length (on the end of the spool when its more bent).
I think the stepper motor has less power as faster it spins, so it get stalled on the high speed but not on the slow one.
in which mode you are using the mmu2 (stealth/normal) ? does it happen in both modes?
is it an upgraded mmu --> mmu2 ?

bynnek, 2020-10-23:
Oh, didn't realize any responded to this!  
I did try going to higher diameter teflon tubes, same issue.  The printer is in normal mode.  Can the MMU be set differently from the printer itself?  I've never tried going to stealth mode. 

This is an upgrade from the original MMU to the MMU2 (and then s) yeah.  Are the old steppers potentially an issue?

bynnek, 2020-10-24:
I think I may have found the problem.  Looks like the male connectors on my rambo are going bad and resistance is increasing.  It appears as though the power plug for the MMU where it connects to the rambo has a very slight sign of melting and the resistance of the connection is higher.  I think I'm not getting enough current to the MMU.  I'll try getting new connectors and bypassing the connector on the rambo and see if that fixes it.

chriswal, 2020-10-24:
Hi, the resistance of the power plug is important, but normally you end up with all leds flashing when there is a problem with it. The plugs should not be melting, there is not enough current used by the mmu2.(if it is there must be a seriuos hardware error)
but it is worth a try. I dont remember why i asked you if it is a upgrade from an MMU . But if it is I think you can swap the two motors (idler / pulley) for a test. this is not possible with the new kit because the pulley motor has also the integrated axle.
what is the type of teflon tube you are using from the mmu2 to the printer. I am using there also a 3mm tube, because sometimes the tips gets wider and get stuck in there.
i dont know if you can change the mode of the mmu2 in the mks2.5 menu .... i have an mk3s and there it is possible.
have you upgraded your printer to 24V or is it a 12V model?