prusa3d/MM-control-01 #172: [Feature] Infer current filament from idler steps to home?

vintagepc, 2020-03-01:
After power cycle if filament is loaded, the MMU does not home and doesn't know what filament is fed. (Common if you leave a filament loaded to nozzle and just slice for a non-MMU printer)

Seems like it should be possible to track how many steps it took (with sensorless homing) to home the idler drum, and back-calculate what filament the selector is currently feeding. 
Would need to keep track of this and do a proper re-home of the selector on the first unload,but it would at least get you out of the state where the MMU wants to unload an unknown filament and is trying to drive the wrong path with the gear.

chriswal, 2020-07-29:
which version you are using? as i remember this works on 1.0.6 and the previous one ..
the mmu stores the last loaded filament in the eeprom
only the idler homes after power cycle (loud rattling) and then moves to the last loaded filament.
the selector is not homed in that state because there is filament in.
if the filament is unloaded als the selector will be homed too.

vintagepc, 2020-07-29:
I'm running 1.0.6.

Perhaps this isn't an issue with the MMU per-se, but the comms between the unit and the printer - such that it could be solved by having a way for the MMU to tell the printer the filament on startup so it does not show "F?".

Bottom line is numerous times I've encountered a situation where an unload in this state fails because the idler is not in the correct location and the retraction fails.

chriswal, 2020-07-29:
yes the printer doesnt know where the filament is loaded. but it should not be possible to move the idler away from the loaded filament unless the finda doesnt detect it anymore.

there is definitly a lot of problems int the error handling of the printer firmware for the mmu2