prusa3d/MM-control-01 #181: Error message when compiling

JohnnyTheOne, 2021-03-05:
when compiling I get this error
`C:\Users\JOHNNY~1\AppData\Local\Temp\ccD3D8hQ.ltrans0.ltrans.o: In function `tmc2130_rx':

sketch/tmc2130.c:320: undefined reference to `tmc2130_cs_low'

sketch/tmc2130.c:321: undefined reference to `spi_txrx'

sketch/tmc2130.c:322: undefined reference to `spi_txrx'

sketch/tmc2130.c:323: undefined reference to `spi_txrx'

sketch/tmc2130.c:324: undefined reference to `spi_txrx'

sketch/tmc2130.c:325: undefined reference to `spi_txrx'

sketch/tmc2130.c:326: undefined reference to `tmc2130_cs_high'

sketch/tmc2130.c:330: undefined reference to `tmc2130_cs_low'

sketch/tmc2130.c:331: undefined reference to `spi_txrx'

sketch/tmc2130.c:333: undefined reference to `spi_txrx'

sketch/tmc2130.c:334: undefined reference to `spi_txrx'

sketch/tmc2130.c:335: undefined reference to `spi_txrx'

sketch/tmc2130.c:336: undefined reference to `spi_txrx'

sketch/tmc2130.c:337: undefined reference to `tmc2130_cs_high'

C:\Users\JOHNNY~1\AppData\Local\Temp\ccD3D8hQ.ltrans0.ltrans.o: In function `tmc2130_tx':

sketch/tmc2130.c:306: undefined reference to `tmc2130_cs_low'

sketch/tmc2130.c:307: undefined reference to `spi_txrx'

sketch/tmc2130.c:308: undefined reference to `spi_txrx'

sketch/tmc2130.c:309: undefined reference to `spi_txrx'

sketch/tmc2130.c:310: undefined reference to `spi_txrx'

sketch/tmc2130.c:311: undefined reference to `spi_txrx'

sketch/tmc2130.c:312: undefined reference to `tmc2130_cs_high'

C:\Users\JOHNNY~1\AppData\Local\Temp\ccD3D8hQ.ltrans0.ltrans.o: In function `tmc2130_init_axis_current_normal':

sketch/tmc2130.c:218: undefined reference to `__res'

C:\Users\JOHNNY~1\AppData\Local\Temp\ccD3D8hQ.ltrans0.ltrans.o: In function `tmc2130_init_axis_current_stealth':

sketch/tmc2130.c:205: undefined reference to `__res'

C:\Users\JOHNNY~1\AppData\Local\Temp\ccD3D8hQ.ltrans0.ltrans.o: In function `tmc2130_init':

sketch/tmc2130.c:247: undefined reference to `selector_step_pin_init'

sketch/tmc2130.c:248: undefined reference to `pulley_step_pin_init'

sketch/tmc2130.c:249: undefined reference to `idler_step_pin_init'

sketch/tmc2130.c:250: undefined reference to `selector_step_pin_reset'

sketch/tmc2130.c:251: undefined reference to `pulley_step_pin_reset'

sketch/tmc2130.c:252: undefined reference to `idler_step_pin_reset'

collect2.exe: error: ld returned 1 exit status

exit status 1
Fout bij het compileren voor board Original Prusa i3 MK3 Multi Material 2.0 upgrade
`

rmilyard, 2021-04-13:
I am getting same now.  You even get it fixed?

JohnnyTheOne, 2021-04-17:
Yes [here](https://github.com/prusa3d/MM-control-01/pull/182)

rmilyard, 2021-04-17:
> Yes [here](https://github.com/prusa3d/MM-control-01/pull/182)

Thanks.  However I am not sure how to use this to fix it.

JohnnyTheOne, 2021-04-17:
Flashing procedure MMU2S [here](https://help.prusa3d.com/en/article/firmware-updating-mmu2s_155118)

[MM-control-01.ino.prusa_mm_control.zip](https://github.com/prusa3d/MM-control-01/files/6329851/MM-control-01.ino.prusa_mm_control.zip)

rmilyard, 2021-04-18:
> Flashing procedure MMU2S [here](https://help.prusa3d.com/en/article/firmware-updating-mmu2s_155118)
> 
> [MM-control-01.ino.prusa_mm_control.zip](https://github.com/prusa3d/MM-control-01/files/6329851/MM-control-01.ino.prusa_mm_control.zip)

Thanks.  However trying to figure out how to compile it on my end.  Some changes I am needing to make.

Alia5, 2021-08-09:
CMakeLists.txt L63: Change `MM-control-01/tmc2130.c`    ->	`MM-control-01/tmc2130.cpp`
Rename 	`tmc2130.cpp` -> `tmc2130.c`
https://github.com/prusa3d/MM-control-01/pull/187