prusa3d/MM-control-01 #183: Is this  MM-control project abandoned?

JohnnyTheOne, 2021-03-16:
Last development activity was 2 years ago.

natthapolvanasrivilai, 2021-03-20:
Might be a 'yes'

ssill2, 2021-04-24:
I was going to ask the same question.

Cx01N, 2021-04-25:
The stability seems to be pretty poor recently and I haven't seen any development in a while. Might have to look at some alternatives.

ssill2, 2021-04-25:
I know in episode #22 of the prusa live youtube show towards the end Josef mentioned an update coming to the mmu firmware.  That was about a month ago.  Maybe they've been working out of a different repo?  Multiple years between commits is not encouraging for sure.

5jvm0u4, 2022-03-27:
I've talked to their customer support and he told me "Not sure if there was any need to improve the current MMU FW version."

ssill2, 2022-03-27:
So that means yes lol.  We all know the current firmware has issues lol.
Too bad

TheDuggem, 2022-03-28:
https://github.com/prusa3d/Prusa-Firmware/issues/1876#issuecomment-1014292539

5jvm0u4, 2022-03-28:
> [prusa3d/Prusa-Firmware#1876 (comment)](https://github.com/prusa3d/Prusa-Firmware/issues/1876#issuecomment-1014292539)

I've saw this one as well, but that reply I got is from last week.

5jvm0u4, 2022-03-28:
[image]

natthapolvanasrivilai, 2022-04-25:
I stop using my MMU because the hardware+software are full of bugs.
The 3D printed parts tolerance can be loosen. The hardware homing can be improved (as seen by TheZeroBeast repo).
The filament feed/retract/eject sequences can be improved and be less prone to minor error.

Pyromane, 2022-09-18:
@leptun @DRracer @michalxfanta
Any Update?

DRracer, 2022-09-20:
@Pyromane MM-control-01 is no longer developed/maintained. 
However, we keep on improving the MMU2S. In the near future, the firmware will be updated. Same goes for the MK3S+ and you can watch the development here https://github.com/prusa3d/Prusa-Firmware/tree/MK3_MMU2.

ggppjj, 2022-09-20:
@DRracer Would it be worth marking this repo as "archived" and leaving that note in the readme? Good to hear, by the way!

DRracer, 2022-09-21:
@ggppjj probably once the new MK3S+MMU FW combo goes out.

ggppjj, 2022-09-29:
@DRracer apologies for the @, looks like that branch was pruned (or at least I'm getting a 404 on that link), is there a better way to keep an eye on progress?

DRracer, 2022-09-29:
the new MMU support has been merged into the main MK3 branch meanwhile: https://github.com/prusa3d/Prusa-Firmware

cjbaar, 2023-01-02:
@DRracer I see a lot of protocol changes in the 3.12 printer firmware related to the MMU, but I don't see anywhere that the actual MMU firmware is being developed publicly to match. Are you able to share this location with us?

leptun, 2023-01-03:
The new mmu firmware is not public yet unfortunately.

cjbaar, 2023-01-03:
That is unfortunate, and seemingly un-open-source-y. :) Is there a way to get beta access?

ggppjj, 2023-01-03:
Likely there will be once the beta firmware is ready for testing 😁